gotthardp/lorawan-server #716: Filter our port=0 frames in Handler or Connector

ahiregange, 2020-04-20:
Hello,

I would like to filter out frames which were received on port=0 from the device and not send it out via Connector/Handler.
Is there a way to achieve this?
The reason is that these frames have empty "data" field and I would like to avoid sending unnecessary and empty HTTP POST.

Thanks,
Arun

altishchenko, 2020-04-20:
You may try setting this up in your handler, 'Parse uplink' field:
```
fun(#{port := 0}, _) ->
    <<>>;
 (Fields, _) ->
    Fields
end.
```

ahiregange, 2020-04-20:
Hi Alexander,

As per your suggestion for an earlier problem I have the following in place to replace the field names of required fields. It is working well. I will try to incorporate your suggestion above for filtering port=0 out. I am still struggling with Erlang. If not too much trouble please suggest how to merge the two.

fun(#{ fcnt := F1, freq := F2, lsnr := F3, rssi := F4, devaddr := EUI , data := PDU } = Fields, _) ->
  #{ <<"SeqNo">> => F1, <<"FREQ">> => F2, <<"SNR">> => F3, <<"RSSI">> => F4, <<"EUI">> => lorawan_utils:binary_to_hex(EUI) , <<"PDU">> => lorawan_utils:binary_to_hex(PDU) }
end.

Thanks,
Arun

ahiregange, 2020-04-20:
Trying this now:

fun(#{port := 0}, _) ->
    <<>>;
(#{ fcnt := F1, freq := F2, lsnr := F3, rssi := F4, devaddr := EUI , data := PDU } = Fields, _) ->
  #{ <<"SeqNo">> => F1, <<"FREQ">> => F2, <<"SNR">> => F3, <<"RSSI">> => F4, <<"EUI">> => lorawan_utils:binary_to_hex(EUI) , <<"PDU">> => lorawan_utils:binary_to_hex(PDU) }
end.

ahiregange, 2020-04-20:
I was able to save the above script in the Handler but it does not seem to have helped to filter out port=0 cases.

altishchenko, 2020-04-20:
Ok, I just don't remember 0.6 line correctly, try returning an empty map `#{}` instead of `<<>>` or empty list `[]`

ahiregange, 2020-04-21:
Sorry, this was my mistake. I had to also add port to list of Uplink Fields of course.
Now it filters correctly.

closed 2020-04-21 by ahiregange