gotthardp/lorawan-server #717: Unable to add new Profiles on the web application v 0.6.7

HMMalek, 2020-04-23:
I want to add a Profile , the create page is not found and returns a 404 error. All other pages on the web application are working and I have already added a gateway, it's working correctly, however I cannot add the Area field that was not included in the version 0.5 and is supposedly necessary for the gateway creation on this version (no attribute with name Area to add in the create page of the gateway) (was using it on a different project; all good).
Anyone faced this issue before? Is there another way to add a Profile ?

altishchenko, 2020-04-23:
If you accessed the 0.5 version of the server from this same browser, it would be helpful to drop the browser cache. This is one of the reasons you don't see the Area field.
As a follow up, I'd recommend upgrading to 0.7.x

HMMalek, 2020-04-24:
Clearing the browser cache worked perfectly. Thank you.

closed 2020-04-24 by HMMalek