gotthardp/lorawan-server #719: Any plans for arm docker container (raspberry)

MihaMarkic, 2020-05-02:
Hi guys,

Are there any plans to build an ARM docker container (for raspberry).
It should be doable, right, I mean, this thing can run on raspberry?

gotthardp, 2020-05-02:
Interesting. It should be doable, i just wonder what the performance penalty will be.

altishchenko, 2020-05-02:
@gotthardp Not a lot, even for the old ones. Can test it on mine. Have you seen my PR?

altishchenko, 2020-05-02:
@MihaMarkic' Hello Miha, I had a quick look around Docker Hub and don't see any problems building a Docker image. I used the server on a raspberry for some time as a tech show equipment, didn't have any problems there. I'll give it a try tomorrow and let you know - have to find my raspberry to test docker on it.

MihaMarkic, 2020-05-03:
@altishchenko Great. Let me know whether I can help (have no experience with this lorawan-server yet, though).

MihaMarkic, 2020-05-03:
@gotthardp I have no idea but I also suspect perf shouldn't be a problem just because of docker.

altishchenko, 2020-05-03:
@MihaMarkic Can you drop an e-mail to atishch@gmail.com? Can't find my raspberry, but you can try something for me.

altishchenko, 2020-05-03:
@MihaMarkic Ok, I found my Pi, it is a 3B (no plus, so arm32v7) model. You can get the image from docker hub, as per the doc/Docker.md, but from another repository:
```
docker pull altishchenko/lorawan-server:armhf
docker run --detach \
  --name lorawan \
  --hostname lorawan \
  --restart on-failure \
  --volume /path/to/local:/storage \
  --publish 8080:8080/tcp \
  --publish 8443:8443/tcp \
  --publish 1680:1680/udp \
  altishchenko/lorawan-server:latest
```

altishchenko, 2020-05-03:
@gotthardp Petr, whenever you will have some spare time, drop me a note or just republish the image to your own repository. (I didn't figure out how to keep same tags for different architectures yet).

gotthardp, 2020-05-03:
@altishchenko Shall we also update something in the documentation? Shall I just rebuild the last stable image, or shall the Dockerfile change somehow in order to include the ARM architecture?

altishchenko, 2020-05-03:
@gotthardp Petr, I am not a Docker guru  yet and trying to figure out how architecture tags work. For now, I just took the original Dockerfile (optimizing PR applied) and built the image on my RPi3B (32bit) without any problems - architecture for the alpine was picked up automatically, then just pushed the result into my repository with tag :armhf. For the moment, I think, it will suffice if you pull this image and push it to yours with the same :armhf tag, describing in the docs the difference between the two (latest vs armhf). Meanwhile, we will figure out how to do proper architecture tags and properly re-publish the images later. May be volunteer for the official images library - this sort of builds/helps to build all the latest architectures and tags automatically.. Can you look into it? And I will dig docker books around me.

gotthardp, 2020-05-03:
I am not an expert either, but i found a possible solution using GitHub Actions. I will try it: https://www.henry.wang/2019/12/05/arm-dockerhub.html

altishchenko, 2020-05-03:
Ok! Keep me in the loop, I am investigating the new manifest lists feature that was added to docker to support multi-architecture multi-manifest repositories. I'll let you know. when I have a working solution.

altishchenko, 2020-05-03:
@gotthardp Namely this one: https://docs.docker.com/engine/reference/commandline/manifest/

altishchenko, 2020-05-03:
@gotthardp Is your centrum e-mail valid?

gotthardp, 2020-05-04:
@altishchenko I made the auto-build working. Github now publishes mult-arch images to DockerHub. The platform is announced as "arm/v7", but it's part of the standard "latest" image. Could you please try if that works with your raspberry too?

altishchenko, 2020-05-04:
@gotthardp Will do that tomorrow. Is your centrum e-mail valid? (And what do you use to build arm64/v8?)

altishchenko, 2020-05-05:
@gotthardp Hello Petr, I verified the setup. Works fine, the image is pulled as expected - linux/arm.

altishchenko, 2020-05-05:
```
$ docker manifest inspect gotthardp/lorawan-server
...
         "platform": {
            "architecture": "amd64",
            "os": "linux"
         }
...
        "platform": {
            "architecture": "arm",
            "os": "linux",
            "variant": "v7"
         }
...
         "platform": {
            "architecture": "arm64",
            "os": "linux"
         }
...
$ docker image inspect gotthardp/lorawan-server
...
        "Architecture": "arm",
        "Os": "linux",
        "Size": 74502803,
        "VirtualSize": 74502803,
...
$ 
```

gotthardp, 2020-05-05:
Cool, then the case is closed :)

Thanks for your help! (BTW, my e-mail works fine)

closed 2020-05-05 by gotthardp

MihaMarkic, 2020-05-07:
Awesome guys, thanks. I'll test it once I figure the basics (on desktop) out.