gotthardp/lorawan-server #720: Minor: Adding a gateway when no areas are configured results in error

MihaMarkic, 2020-05-07:
If there are no areas configured then adding a gateway yields "Invalid Form", but area input field doesn't get colored red nor there is a useful error description.
So a bit misleading for newbies.

altishchenko, 2020-05-07:
Yeah, the web interface was originally intended as a demo only :). The errors are not helpful there 😄

gotthardp, 2020-05-07:
Yeah, the framework used for web admin is sometimes buggy and even worse, it is not maintained anymore. We are supposed to migrate the entire web to React, but it is not as easy as I thought.

MihaMarkic, 2020-05-08:
React is easy, easier that classic stuff. Once you understand it and get some practice, that is. For me a good way to learn it was a course on Pluralsight.

altishchenko, 2020-05-08:
@MihaMarkic Can you possibly volunteer? Such help would be greatly appreciated...

MihaMarkic, 2020-05-08:
@altishchenko Honestly, I'd like to. Realistically it's really tough time for this for me. There is another approach perhaps, that's more appealing to me these days and I'd try to do something with it - Blazor. It's .net stuff in some ways similar to how react works. Either way my free time is really a problem these days. I'll think about it, but I'd prefer Blazor.