gotthardp/lorawan-server #723: About AppEUI, DevEUI, AppKey and endian

MihaMarkic, 2020-05-13:
Have two questions regarding About AppEUI, DevEUI and AppKey.
1. What is the endian case in web ui? Big or little? I guess big.
2. How does one generate AppEUI and AppKey? DevEUI is fixed by manufacturer, right? I guess AppKey can be a random bunch of bytes, but what about AppEUI?

Thanks a lot.

gotthardp, 2020-05-13:
1. The GUI follows the LoRaWAN standard. It's a big endian.

2. DevEUI is set by manufacturer. Some part (first 3 bytes, OUI) is fixed for each OEM, the rest identifies a device. Could be a sequence, or random. Your choice.
AppKey is a secret. Could be a random, or a hash. Your choice.
AppEUI is an identifier that isn't really used in LoRaWAN 1.0 much. Use whatever number you like.

altishchenko, 2020-05-13:
@MihaMarkic To put things into perspective a bit, in the world of LoRaWAN 1.0.3 AppEUI identifies the "application" running on your end device. It is used for identifying the data and actions your device is capable of, so the network administrator or service provider can choose the appropriate high-end application to receive and process your device's data. At least this was an intention. Now at the days of LoRaWAN 1.1 the focus has moved towards security and splitting roles and authority. In 1.1 speak these fields are now called JoinEUI and JoinKey, so you can guess from that, that these two fields identify the entity responsible for authenticating your device on the network - allowing it to join and establish the session (keys).
With this server you shouldn't bother with all that, just make sure the stuff on the device matches the server settings.

MihaMarkic, 2020-05-14:
Thanks @altishchenko and @gotthardp

closed 2020-05-15 by MihaMarkic