gotthardp/lorawan-server #724: Buggy Administrator web and alternative methods

Gillani0, 2020-05-15:
Hi,

There seems to be many issues while creating the new network and device groups (invalid forms) and sometimes submitting the form doesn't work. I used the stable releases. Furthermore, when I build the sources and used it, the web page doesn't open due to some React issues. 

Therefore, is there any other alternative way of configuring networks and devices (than HTTP REST)? If not can you please direct me to the complete documentation of the REST API when request and responses etc. Cheers

Syed

gotthardp, 2020-05-15:
Hi. The stable versions have Angular only. It is not so buggy. After upgrade the browser cache should be cleared. Other than that, it should work. There is just the documentation in the doc folder. It is not super-complete, but quite simple so the API fields shall be self-explanatory.
Contributions are welcome to complete the React admin, which I hope will be better.

Gillani0, 2020-05-15:
Thanks for the quick response, but I've tried the stable versions with many browsers but it's unable to send the Post request to create a network (pressing the button doesn't do anything). 
The documentation in the doc doesn't describe the request structure :) 

I'll probably have to dig in the code to check this. 

Syed

gotthardp, 2020-05-15:
Have a look in the browsers debzg console. Is there a javascript error? Is a http request sent?
I suspect you might not filled all mandatory fields.

Gillani0, 2020-05-15:
Yeah, I've debugged it, no HTTP request is sent and all the fields on the form are marked as green.

altishchenko, 2020-05-15:
Hello @Gillani0 , I will also strongly advise clearing the browser cache, I was using the web-interface for many different versions of this server and usually cached .js files was the problem if anything went wrong. Also, are you connecting to the server via any sort of proxy? Or, possibly, you can download and run locally a servers docker image to check?

Gillani0, 2020-05-16:
Hi @altishchenko, thanks for the tip. I'm running the server locally (so no proxy or firewall etc). The issue is not the request being blocked, they are just not generated when the button is clicked to send the Post request. I'm actually quite baffled by this issue and have cache was cleared many times and I've used many different browsers.  Cheers

Gillani0, 2020-05-20:
So, in the end, the issue was more of a presentation than the functionality. 

In the new network creation, there are multiple tabs each with a submit button, the new network submits button is quite misleading. All the info in the other tabs has to be first filled and only then you should click the submit button to it can send the request.

closed 2020-05-20 by Gillani0

altishchenko, 2020-05-20:
Oh sorry, I didn't take that into account. Somewhat out of habit I am filling in all the data on all tabs and press submit once... The same applies to other multi-tab Edit dialogues in the server.