gotthardp/lorawan-server #725: Filter connector's forwarding by node's devAddr

arunaseva, 2020-05-15:
Hi,

I have a working installation with version 0.6.7.

I created a handler and a connector:
- The handler's uplink fields: devaddr, data, ...   (Payload: ASCII Text)
- The connector send uplink messages to an MQTT server's topic in JSON format.

I want to filter nodes by device address, so I don't want to forward all node's messages by the connector to the MQTT broker, just some selected with specific devArr.

Is it possible to set up this in the web admin or any other solution?

Thanks, Adam

gotthardp, 2020-05-15:
This could be solved by assigning the app/handler to some nodes only.

altishchenko, 2020-05-15:
@arunaseva Hello Adam, the right way to do this will be (you did most of it, but some steps remain):
1. Create a Handler to process messages (note, ASCII text should be used if your device really sends just ASCII strings).
2. Create Connector of choice and attach it to the Handler.
3. Create a Profile for your devices and assign created Handler (Application) to it.
4. Create your devices and assign them to this Profile if you want these devices to use your connector.

In your setup you need steps 3 and 4 - assigning different devices to different Profiles is how separation is done.
device1 -> Profile1 -> Handler1 -> Connector(s)-1
device2-> Profile2 -> Handler2 -> Connector(s)-2

arunaseva, 2020-05-19:
"Profile for your devices and assign created Handler (Application)"

Thanks a lot, entirely clear now.

closed 2020-05-19 by arunaseva