gotthardp/lorawan-server #726: Many ack_lost, uplinks_missed and some repeated_requests

MihaMarkic, 2020-05-15:
Hi guys,

I've finally set up a single configuration consisting of a device, gateway and this server. Thanks to guys here.
Anyway, while my device works - is able to send data and server receives it, I see plenty of ack_lost and uplinks_missed warnings. My gateway is [ESP-1ch-Gateway](https://github.com/things4u/ESP-1ch-Gateway) and my issue there is https://github.com/things4u/ESP-1ch-Gateway/issues/34.
From what I read here and in troubleshooting, the problem is with gateway and it's setting _push_timeout_ms_.

gotthardp, 2020-05-15:
Ou, is that a single-channel gateway? Those things aren't very standard-compliant and don't work with the default config. You need to convince both the server and the device to use that single channel the gateway has.

MihaMarkic, 2020-05-19:
Yep, single channel. The thing is that the data comes through from the device, it's just those warnings for some reason.
Do you happen to know if there is an alternative gateway that would run on a TTGO ESP32 board?

gotthardp, 2020-05-19:
Just of out curiosity, could you check the Frames page to see if there are any FCnt gaps? Devices do frequency hoping, so a single channel gateway receives only some uplinks. Could that be a reason for the errors?
BTW, I no almost nothing about the gateway market.

michello-zz, 2020-05-23:
Hello good evening, I implemented a gateway (single channel) with TTGO and it works very well in the uplinks but I couldn't do the downlinks.
I use the LMIC library and a node with another TTGO.
From now, thank you for you help

altishchenko, 2020-05-23:
@michello-zz Should be no problems there, what setup do you use?  Which variant of LMIC do you run, mcci or kooijman?

michello-zz, 2020-05-25:
Thank you very much for your answer.
I am using an adaptation for my zone "Argentina" AU915 915 MHz to 930 MHz.
https://github.com/eMGing/LMIC_Shield_eMGing_LoRa_Node_915_LoRaWAN
  * http://www.eclipse.org/legal/epl-v10.html
  *
  * Contributors:
  * IBM Zurich Research Lab - initial API, implementation and documentation
Node: TTGO
Gateway: Mono Channel TTGO
Thank you in advance for your help.
Greetings Michello

altishchenko, 2020-05-25:
@michello-zz Ok, this library is based on this source: https://github.com/matthijskooijman/arduino-lmic, can you share your server configuration and also provide some details on how are you trying to do the downlinks? Are your uplinks confirmed or unconfirmed?

michello-zz, 2020-05-25:
Yes of course
attached files 
[ESP-sc-gway.txt](https://github.com/gotthardp/lorawan-server/files/4679137/ESP-sc-gway.txt)
[Nodottn-abp.txt](https://github.com/gotthardp/lorawan-server/files/4679139/Nodottn-abp.txt)
[ESP-sc-gway.h.txt](https://github.com/gotthardp/lorawan-server/files/4679148/ESP-sc-gway.h.txt)
[ESP-sc-gway.txt](https://github.com/gotthardp/lorawan-server/files/4679149/ESP-sc-gway.txt)
[config.h.txt](https://github.com/gotthardp/lorawan-server/files/4679150/config.h.txt)
[lmic.c.txt](https://github.com/gotthardp/lorawan-server/files/4679151/lmic.c.txt)
[lmic.h.txt](https://github.com/gotthardp/lorawan-server/files/4679152/lmic.h.txt)

michello-zz, 2020-05-25:
How do I get the server configuration

altishchenko, 2020-05-26:
@michello-zz Well, again:
How are you trying to send downlinks?
What commands do you use for that?
Can you show us screenshots of your server's web-admin: particular Handler and Connector configuration panels, the ones used by your device (device's profile)?
Can you try sending a downlink message and after waiting for some time (your device should send an uplink towards the server at least) capture the debug.log file from the server and share it too?

P.S.: You can always export complete configuration for your server with dbexport command, found in /your/path/to/installation/bin/ directory. For example, if you installed your server on some sort of debian or ubuntu you will normally find it in /usr/lib/lorawan-server/bin and the debug.log file should be located in the /var/log/lorawan-server/ directory.

michello-zz, 2020-05-26:
Hello good day.
I use MQTT for the "downlinks".
Attached configurations and log.
Attached captures the frames, where the downlinks are seen.
[backup-2020-05-22.zip](https://github.com/gotthardp/lorawan-server/files/4682171/backup-2020-05-22.zip)
[debug.log](https://github.com/gotthardp/lorawan-server/files/4682158/debug.log)
[image]
[image]
[image]
[image]
[image]
[image]
[image]

altishchenko, 2020-05-26:
@michello-zz Ok, I had a look around in your logs and screenshots, I didn't look into your configs yet. There are a few things that really worries me:
1. Your device sends periodic garbage, which means that either there is not enough memory on the stack or your data sending routine pollutes the transmit buffer or your gateway has errors of the same type.
2. Surprisingly, though I presume the timestamps are correct, I did not see any indication in the server log that data was received via MQTT and was transmitted to your device.

There are two things you can try:
1. Change the Format field in your connector to from RAW to JSON, from the frames screen shot I can see that the server transmits literally '{"data"..}' string to the device. I think this is not what you want.
2. To make things easier to debug, create a new Connector with URI `http:` - just that and 'Received topic' to `/down/{devaddr}` and attach it to the same Handler/Application. This will enable you to send downlinks with usual curl/http command:

```
curl -v --digest -uadmin:admin -H "Content-Type: application/json" -X POST http://localhost:8080/down/11223344 -d "{\"data\":\"ABCDEF\"}"
```

Clear the log, ex. stop the server, delete logs, start the server. Then send a few frames down to the device with the command above, make sure the device sends a few frames up to receive them.
And send me the log to atishch@gmail.com, as this issue is unconnected to your problem absolutely.