gotthardp/lorawan-server #727: Details about the Join Mechanism

Gillani0, 2020-05-21:
Hi,

I wanted to understand the join mechanism that's implemented for this network server. According to my knowledge (considering the **Class A device** and **Semtech UDP packet forwarder**). 

1- When the `Join Request` is received if keys matched, etc, the `Join Accept` response will be queued for the `Join Delay 1` using the `Rx1` configuration. 
2- With the next `PULL_REQ` requests from the UDP packet forwarder the queue will be continuously checked and when the delay will be about to be expired the `Join Accept` will be sent to the gateway to be sent on the radio. 

However, from my test, it seems like:
1- The `Join Accept` response is sent as soon as the `Join Request` is validated.
2- The server doesn't wait for the next `PULL_REQ` from the packet forwarder.  
3- The `Rx1` and `Rx2` configuration` are alternated, i.e. for the first `Join Request` it uses `Rx1` and for the second `Rx2` for the third again on `Rx1` so on and so forth. 

Is this correct or am I missing something? Cheers

gotthardp, 2020-05-21:
Hello. I believe the server behaviour is correct:
1) The TX queue is in the gateway, so the server sends a downlink to the gateway immediately and the frame is stored there until the exact scheduled time.
2) The server is free to send the PULL_RESP "anytime after first PULL_DATA".
3) The alternating is an additional intelligence. The server is free to use either TX1 or TX2. I implement the altenation of retransmissions (only!) to recover from errors and misconfigurations because of the frequencies could be blocked.

Gillani0, 2020-05-21:
Thanks for your quick response.
1. From this, does it means that the server sends the downlink after the `PUSH_DATA` even if no `PULL_DATA` is received?
2. I agree (https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT) it should send it after the `PULL_DATA` but in my tests, it sends (for the Join requests) even if no `PULL_DATA` is received.   

Further exploring the first point, thanks for explaining that you expect the gateway to store and send the message at the scheduled time. However, there are many gateways that don’t have enough memory to fit more than one scheduled message. So maybe in the future, you can add a queue at the network server-side instead of relying on the gateway to handle this.

altishchenko, 2020-05-23:
@Gillani0 The protocol says "after the first PULL_DATA for every packet to TX" and it may be any pull data that didn't time out after the server marked the gateway as connected, it is not bound to a particular device or a particular device's packet received. This mechanism is mainly used to solve the problems of the gateways behind NAT firewalls, where connection path in the firewall exists for a short amount of time while there is communication going on. It is a mechanism for the gateway to keep this path open in case the server has something to send back.
As for the second issue: if the gateway **does not** implement buffering until the exact transmission time happens it is violating the LoRaWAN standard and cannot be called a gateway in a true sense. The Network server is incapable of keeping the exact transmission times because of the obvious network latency and delay issues and unpredictable network behaviour, network server cannot by any means guarantee downlink frame delivery hitting the **exact** RX time window, it is the gateway's job sorry.
There is only one queue and it is implemented: downlink frames are queued to class A devices waiting for the node to provide an uplink and open an RX window.

Gillani0, 2020-05-26:
@altishchenko  Thanks for your explanation. My understanding was that the gateway has to open the connection before getting any response (due to the inherent firewalls etc). But yeah you're correct, we can send multiple Tx messages after the first `PULL_DATA`.

And yeah I agree with the second part as well (after looking at the code of packet forwarder). Cheers

closed 2020-05-26 by Gillani0