gotthardp/lorawan-server #730: Event propagation for unknown_deveui

fisch111, 2020-05-28:
Hi,
I'm testing your wonderful LoraWAN server for the purpose of using it in a training for our customers. So far so good, except for simplicity reasons I would need to register the devices automatically, without filling out each form for commisioning. I know there is REST api for that, but now I'm stuck with the idea of propagating the error unknown_deveui event to my Node Red application level and from there sending the API request to register the device. I guess it is because this event is marked as error event, which should not be propagated, so is there a way to change it to info event and get it to my app server? Any other info event like join is successfully propagated via MQTT to the app server, so the connection is not an issue. Or is there another way how to automatically register a device? 
Thank you in advance.

Best regards,
fisch111, a fellow engineer from Prague :)

gotthardp, 2020-05-28:
Ahoj to Prague. Perhaps the easiest and cleanest way to have an auto-registration via a script would be to allow for an external "registration service" that would receive a REST API call when some external data are needed. I wouldn't combine it with the MQTT interface, which is mainly indended for frames.

fisch111, 2020-05-28:
Ahoj,
thank you for the quick reply. Sorry for the buzz, I'm a embedded engineer, so these sw things are a bit new to me. I had a MQTT connection already, so I wanted to use the same source. But it seems to create a simple flow in Node Red to GET the event and then create the device with another request is the way to go :). Hopefully I will handle it from Node Red, personally I would prefer Node.js directly, but again for simplicity reasons I need to use Node Red.

Best regards,
fisch111

altishchenko, 2020-05-28:
Hello guys, just my five cents.
We tried implementing a sort of a 'proper' external registry (or Join server in 1.1 speak), but it in the end it all became rather complicated - too many things to keep in sync between the network server and the registrar, so we ended up with a custom event/AMQP message, that will trigger certain hooks in our app which will in turn deploy a required config back on the network server via REST API. It isounds like a mess and it is. Sorry Petr, I didn't post this code for consideration as it really is a dirty hack and I hadn't the time to do it the right way.
I'd rather go now with a generic server event mechanism, that is configured on the server config level more or less like the existing unused slack channel, that will provide server level events (our dear :throw_event() ones will be great) via a configurable method, not bound to the usual profile-handler-connector chain. For this event, things like serverid (erlang node), gw data, area could be of great help where present, but not really required, because this can complicate things too much.
Petr, what do you think of extending throw_event: with such notification?

altishchenko, 2020-06-02:
@fisch111 I was experimenting with event notifications and came up with some preliminary solution for you. For the moment as the implementation is not final and few roughs remain, you can get the code from my fork's events branch. If you require a docker image, let me know and I'll build you one (non-master branches are not built automatically into images). How it works:
1. Pick a connector of your choice and configure its `Publish Events` topic. Make sure you **don't** use any specific macro expansions like {devaddr}, {appid} in it.
2. Go to the Server -> Configuration tab.
3. If the line `Send events to` doesn't show up, forcefully refresh the page (Shift-F5 for Chrome).
4. From the fall down list choose an Application handler to which your Connector from 1. is attached.
5. Start listening to the topic configured in 1.

You will start getting all sorts of messages on that topic, some of them **may** happen invalid for MQTT. Please, let me know of those messages, because I am testing with AMQP broker which is similar, but different.

For the unknown devices you will get a message like this (eid is the deveui in this case):

```
{"app":"Eventer","datetime":"2020-06-02T00:49:12Z","eid":"0A0D477361394855","entity":"device","event":"unknown_deveui","severity":"error","sname":"lorawan@cwp-loradev"}
```

altishchenko, 2020-06-09:
@gotthardp, @fisch111 Guys, is this topic of any interest to you?
Petr, I have a working variant and can do a PR if you think this might be helpful, would you like to make a review?

commit 1a5c0f2f (2020-11-15): Ability to select one handler as a global event receiver (resolves #730) (#758)

Publish generic events via a standard Handler connected to the server config globally.
closed 2020-11-15 by gotthardp