gotthardp/lorawan-server #735: Downlinks visible on Node but not on Frames

ahiregange, 2020-06-18:
Hi,

I went through https://github.com/gotthardp/lorawan-server/blob/master/doc/Connectors.md#demo-page to understand how to send downlinks. I am using v0.6.7.

As I already have an Uplink Connector in place to forward data to our cloud, I concluded from reading this that I could simply add /in/{devaddr} to Received Topic and then send a curl command as mentioned:

curl -v --digest -uadmin:admin -H "Content-Type: application/json" -X POST http://lora-staging.iotini.com:8080/in/10000110 -d "{\"data\":\"01010101\"}"

The good news is that this seems to be successful because as soon as I do this I can see the Downlink queued on the Node page and with the same data as above. Please see attached image.
[image]
After a few minutes i.e. I think when the next Uplink happens, this goes away, which seems to indicate the queued Downlink got sent.

However, a few observations & questions:
1) I do not see the same data on the Frames page, which I would have expected if it was indeed sent Downlink. Is this expected to be visible there once sent?
2) I also checked on the LoRa Gateway and check if any Downlink payload was logged but I don't see it. I do see the Downlink Ack (with 0-size payload) on the gateway after every Uplink.
3) I have pasted after this message the full log for the curl command, I see a couple of HTTP error codes. I do not know if these are relevent as the data does reach the LoRaWan server as mentioned above.
< HTTP/1.1 401 Unauthorized
< HTTP/1.1 204 No Content
4) I also noted as pasted in above image of Node page that the Port field is empty for the Downlink. This may be the key to the issue? What should I set it to in the Curl command?
5) What is the format of the data? In the documentation it mentions "{\"data\":\"ABCDEFG\"}" : is this ASCII text or raw hex data? I need to send raw Hex data and not ASCII.

Thanks a lot!

Arun

==========================================================
*   Trying 45.55.164.111...
* Connected to lora-staging.iotini.com (45.55.164.111) port 8080 (#0)
* Server auth using Digest with user 'admin'
> POST /in/10000110 HTTP/1.1
> Host: lora-staging.iotini.com:8080
> User-Agent: curl/7.47.0
> Accept: */*
> Content-Type: application/json
> Content-Length: 0
>
< HTTP/1.1 401 Unauthorized
< content-length: 0
< date: Thu, 18 Jun 2020 10:24:59 GMT
< server: Cowboy
< www-authenticate: Digest realm="lorawan-server", nonce="bf6495a214b8bbd6f41ab9a37bc1236b", domain="/"
<
* Connection #0 to host lora-staging.iotini.com left intact
* Issue another request to this URL: 'http://lora-staging.iotini.com:8080/in/10000110'
* Found bundle for host lora-staging.iotini.com: 0x55672ee5a4d0 [can pipeline]
* Re-using existing connection! (#0) with host lora-staging.iotini.com
* Connected to lora-staging.iotini.com (45.55.164.111) port 8080 (#0)
* Server auth using Digest with user 'admin'
> POST /in/10000110 HTTP/1.1
> Host: lora-staging.iotini.com:8080
> Authorization: Digest username="admin", realm="lorawan-server", nonce="bf6495a214b8bbd6f41ab9a37bc1236b", uri="/in/10000110", response="135b9568cc8622a099aec28fcbd5173d"
> User-Agent: curl/7.47.0
> Accept: */*
> Content-Type: application/json
> Content-Length: 19
>
* upload completely sent off: 19 out of 19 bytes
< HTTP/1.1 204 No Content
< content-type: text/html
< date: Thu, 18 Jun 2020 10:24:59 GMT
< server: Cowboy
<
* Connection #0 to host lora-staging.iotini.com left intact
==========================================================

ahiregange, 2020-06-18:
Hi,

Adding here the filtered downlink frames after the above downlink was queued in case it helps you.
[image]

Thanks,
Arun

ahiregange, 2020-06-18:
Hi,

I added "port":2 to the curl command and I see the port listed as 2 for the Node Downlink.
But as seen in the attached image, the following downlinks still do not show this port nor data.
[image]

Thanks,
Arun

altishchenko, 2020-06-18:
Do you see any other errors on the dashboard or in the logs? Look into the debug.log, there must be some indication.
As for your other questions: the data you send is a HEX string, it gets decoded on the server and is sent as binary to your device.
Also, can you show me your connector configuration and your curl command?

altishchenko, 2020-06-18:
At the first looks your curl command is a little wrong, you should use single quotes around the message after -d, like this:
```bash
curl -v --digest -u admin:admin -X POST http://yoursever:8080 -d '{"data":"01020304"}'
```

ahiregange, 2020-06-18:
Hi Alexander,

I have taken the curl command directly from the server documentation i.e. please see this snip from there:
[image]

Also the data does reach the server on the Node page Downlink section as shown here:
[image]
But somehow it does not make it to the Frames page nor to the gateway.

1) I will anyway try with single quotes as you suggested.
2) And I will look at debug.log

Thanks,
Arun

ahiregange, 2020-06-18:
With single quotes I get a different HTTP error:
< HTTP/1.1 400 Bad Request

Also the dashboard shows a red box with "json_syntax_error" and yellow box with the HTTP 400 error, if I use single quotes. So I think double quotes as given in your documentation is correct.

ahiregange, 2020-06-18:
Hi Alexander,

I do see errors in debug.log, exactly when the queued downlink was going to be scheduled. Just to clarify, the downlink reaches lorawan server fine and is queued fine. These are the relevant lines in debug.log:

2020-06-18 15:05:35.219 [warning] <0.28012.97>@lorawan_connector_http:terminate:147 Connector Cloud.Iotini.Connector terminated: {function_clause,[{lorawan_admin,build,[<<>>],[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_admin.erl"},{line,142}]},{lorawan_connector_http,encode_uplink,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_http.erl"},{line,235}]},{lorawan_connector_http,handle_uplink,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_http.erl"},{line,200}]},{lorawan_connector_http,handle_info,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_http.erl"},{line,68}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
2020-06-18 15:05:35.219 [error] <0.28012.97>@lorawan_admin:build:142 gen_server <0.28012.97> terminated with reason: no function clause matching lorawan_admin:build(<<>>) line 142
2020-06-18 15:05:35.219 [error] <0.28012.97>@lorawan_admin:build:142 CRASH REPORT Process <0.28012.97> with 0 neighbours crashed with reason: no function clause matching lorawan_admin:build(<<>>) line 142
2020-06-18 15:05:35.220 [error] <0.14629.52>@lorawan_admin:build:142 Supervisor lorawan_connector_sup had child <<"Cloud.Iotini.Connector">> started with lorawan_connector_http:start_link({connector,<<"Cloud.Iotini.Connector">>,<<"Cloud.Iotini.Handler2">>,<<"json">>,<<"http://cloud.iot...">>,...}) at <0.28012.97> exit with reason no function clause matching lorawan_admin:build(<<>>) line 142 in context child_terminated
2020-06-18 15:05:35.221 [debug] <0.28448.97>@lorawan_connector_http:ensure_gun:160 Connecting Cloud.Iotini.Connector to http://cloud.iotini.com:10111

Thanks,
Arun

ahiregange, 2020-06-18:
This is my connector configuration:
[image]

altishchenko, 2020-06-18:
I see, you didn't write about backslashes in the fist post. To use single quotes you must remove backslashes.
Now, I see the error. Can you post your application handler config? It looks like there are options missing.

ahiregange, 2020-06-18:
This is my Handler configuration, which is basically blank for Downlink as I do not need any special handling. Is this okay?

[image]

Based on this documentation I left it blank:
[image]

ahiregange, 2020-06-18:
Handler config with Uplink as well. The Uplink part has been working perfectly, in fact it was you helped me to get it to work:

[image]

altishchenko, 2020-06-18:
Ok, everything looks fine...
Let's get rid of the error first. The error is produced by the first part of the Parse Uplink, consider changing it to this:
```erlang
fun(#{port := 0}, _ ) ->
  #{};
```

As for the downlinks, I'd like to have a look at the full debug.log after you queue downlink and it disappears from the server. I can't see any problems in your config.

ahiregange, 2020-06-18:
I tried the change to the Uplink you suggested and it is working, I do not see those lines showing crash etc. in debug.log. Now the much smaller log at exactly same time when Downlink vanished from Node and didn't appear in Frames:

2020-06-18 16:23:56.043 [debug] <0.31557.97>@lorawan_mac:check_fcnt:287 10000110 fcnt reset
2020-06-18 16:23:56.253 [debug] <0.31557.97>@lorawan_mac_commands:build_fopts:77 10000110 <- [dev_status_req]
2020-06-18 16:24:03.370 [debug] <0.31490.97>@lorawan_mac_commands:handle_fopts:16 10000110 -> [{dev_status_ans,0,8}]
2020-06-18 16:24:03.371 [debug] <0.31490.97>@lorawan_mac_commands:handle_status:236 DevStatus: battery 0, margin: 8 (max -15.0)
2020-06-18 16:24:03.441 [debug] <0.31179.97>@lorawan_connector_http:handle_info:117 HTTP request failed: 422, {<<"http://cloud.iotini.com:10111">>,<<"//messages">>}
2020-06-18 16:24:03.441 [warning] <0.31179.97> connector_http {http_error,{422,<<"http://cloud.iotini.com:10111">>,<<"//messages">>}}

ahiregange, 2020-06-18:
I have to check about the HTTP error 422 though. It was not happening previously.

altishchenko, 2020-06-18:
This is port 0 with empty message body (your device sends ADR responses on port 0), try changing empty map #{} to empty list []. Unless suppressing uplinks became broken some time in the past.

altishchenko, 2020-06-18:
What code base do you use for your device? Is it LMIC based?

ahiregange, 2020-06-19:
Hi Alexander,

1. Thanks for the suggestion to use [], now I do not get the HTTP 422 error due to empty data in the port 0 MAC/ADR response.

2. I tried a downlink again after making the above change (just in case) but there is no change i.e. data gets queued at Node Downlink and then vanishes when new Uplinks come in but the data does not show up on Frames. There is no error in debug.log either.

Question: how can I track what is happening from the time the Downlink is queued at Node to when it vanishes?

BTW our device code is based on https://github.com/Lora-net/LoRaMac-node ported to our platform.

Thanks,
Arun

altishchenko, 2020-06-19:
Arun, do you reset your device before or after you queue a downlink to speed things up? I noticed from the logs above that device reset itself:
```
@lorawan_mac:check_fcnt:287 10000110 fcnt reset
```
Upon reset all **previously** queued downlinks are discarded.

altishchenko, 2020-06-19:
Btw, this I my favorite stack, we use it in all our devices. What version are you at? Do you use Lmh or have your own state machine? If the latter is the case, I'd suggest a small modification in SendFrame() functions - not to send an empty frame to push MAC commands and wait until some user data is available - it does not hurt, but you will get rid of port 0 messages altogether.

ahiregange, 2020-06-19:
Hi Alexander,

This is a good point! I didn't think of it. Due to various reasons like reducing power consumption we do start from scratch each time. Sorry, I did not realise this also clears any queued downlink. I will test without this and see if downlink works.

I am currently at v4.4.4 of the device stack. I did try to get rid of the port 0 message, what I saw was that even though I had data to send, SendFrame() sent uplink MAC info and the available data went out only in the next round. But I am okay with this because it sends status periodically due to this which is not bad.

Thanks,
Arun

altishchenko, 2020-06-19:
Ah, ok. I prefer to send device_status_ans MACs with data frames, cause we have plenty of data and this is not an urgent message.

ahiregange, 2020-06-19:
How are you able to control that i.e. send device_status_ans MAC with data frames? I agree it is preferable. Do you have any example code which will help me?
I guess I do not know the stack well enough.

ahiregange, 2020-06-19:
And I just tried initiating a downlink while the device is active i.e. after reset and I immediately see the downlink data in Frames.
Thanks!

altishchenko, 2020-06-19:
What MCU do you use? STMs have a very good Stop mode after which you don't have to reset the connection, for the other MCUs, that can't retain RAM during Sleep/Stop it is much better to consider properly using NvmCtxMgmt set of functions - to save connection state like keys, fcnts, etc in FLASH/EEPROM or in user registers in case of nRF.

ahiregange, 2020-06-19:
I will try to duplicate this result a few more times.

altishchenko, 2020-06-19:
Ok, there is just a special hook in the server code that purges all queued downlinks when the device resets itself... lorawan_mac:reset_node()

ahiregange, 2020-06-19:
We are using Silicon Labs MCU.
I did see the NvmCtxMgmt functionality in new versions but I have not tried it yet.

altishchenko, 2020-06-19:
Oh, it is pretty old really :) I remember it from 4.3.x. For it to work you should have a working flash/eeprom/someother storage solution.

ahiregange, 2020-06-23:
I have changed device side code so that LoRa stack is not re-initialized each time and now queued downlink works. :)
Thanks!

closed 2020-06-23 by ahiregange