gotthardp/lorawan-server #738: Custom devices support

Kralg, 2020-07-08:
Hello,

I am using LoraWan devices with this server and that is good. As there are some other types of devices it would be good to connect additional devices compatible with Lwm2m or other communication to this server. Can you suggest what could be the best option to have a single UI with information from Lorawan devices and Lwm2m (or MQTT, TCP client)? Regarding Lwm2m there is a Eclipse Leshan Lwm2m server demo implementation, it is in java though.

altishchenko, 2020-07-23:
Hello, sorry for the delay with the answer. Petr seems to be unavailable.
Basically, LWM2M and LoRaWAN are close technologies, but utilizing a little different transportation approach and ideology it is not possible to mix the two. It does not sound right to implement lwm2m mechanics on a dedicated LoRaWAN server software given the subtleties in the underlying technology.
For this very reason this server supports AMQP/MQTT broker service - to channel the data and some other control information towards the application services and back to the devices disregarding the underlying media. Using such broker constructs makes it possible to use any technology you want to collect and distribute data while leaving technological specifics to the end-point dedicated servers.

Kralg, 2020-07-29:
Thank you. 
Meanwhile I read about emq x integration for lwm2m. I will try to look forward to couple them together.

closed 2020-07-30 by Kralg