gotthardp/lorawan-server #742: Send email

JorgeLA, 2020-07-14:
Is there any way to specify the port number for send mail? In my case port 25 is closed by the firewall. Thank in advance

jcn, 2020-11-08:
From the looks of #523 it seems like there is no way of changing the smtp port this at this time.

altishchenko, 2020-11-08:
This can be easily implemented, if there is a need.

JorgeLA, 2020-11-08:
Can you share how to implement it?

altishchenko, 2020-11-08:
@JorgeLA There are two ways to do it: 1) quick and dirty, 2) slow, but right.
Quick and dirty can be done with any text editor:

1. In the lorawan_server.config file add a line in the `lorawan_server` section with `{ smtp_port, 8225},`

```erlang
{lorawan_server, [
    ...
    {http_admin_credentials, {<<"admin">>, <<"admin">>}},
    {smtp_port, 8225},
    ...
```

2. In the lorawan_db_guard.erl file, in the function `send_emails0`, change two lines 284-285 to this:

```erlang
    Port = application:get_env(lorawan_server, smtp_port, 25),
    gen_smtp_client:send({From, ToAddrs, mimemail:encode(Body)},
        [{relay, Server}, {port, Port}] ++
```

3. Recompile and install.

The *slow, but right* way involves changes to admin.js and server configuration database, apart from changes to the code. This can be done too, if really required, but I can't properly explain how in a message.

jcn, 2020-11-08:
@altishchenko with the "slow, but right way," are you describing actually updating the interface to add a port to the same configuration that the server name, username, and password are added?

I ask specifically because I run the `lorawan-server` out of Docker, which is super convenient, but that also means that I'm not going to be recompiling the server any time soon (and I suspect most users of this package probably won't be either).

altishchenko, 2020-11-09:
@jcn Yes, that is what I mean. But, not to worry I keep "devel" dockers in my docker account, while Petr keeps releases and the top of the master branch.

jcn, 2020-11-09:
@altishchenko Sorry, specifically I mean: since I run the server out of Docker, then there really isn't a practical way for me to change the SMTP port right now, right? Thanks!

altishchenko, 2020-11-09:
@jcn Right now, yes. (but you can patch the image of course ;) ). If there is a need, I can implement this option in my devel fork, docker images are available at [my fork's repository](https://hub.docker.com/repository/docker/altishchenko/lorawan-server).