gotthardp/lorawan-server #745: ** URGENT ** I cant restart LoRaserver, the console windows crashes LOG-files attached

hjojohan, 2020-09-03:
Hi 
I cant restart Lorawan server 0.6.7, i tried to right click and restart .bat file as admin.. the console window crashes. Can anyone spot whats wrong from the attached log files ?  Erlang is version  OTP22.  

[debug.log](https://github.com/gotthardp/lorawan-server/files/5167123/debug.log)
[error.log](https://github.com/gotthardp/lorawan-server/files/5167126/error.log)
[crash.log](https://github.com/gotthardp/lorawan-server/files/5167127/crash.log)

gotthardp, 2020-09-03:
Hello. This looks like a database or file corruption. You may need to reinstall and restore the database from backup.
 
Petr

> Od: "hjojohan" <notifications@github.com>
> Komu: "gotthardp/lorawan-server" <lorawan-server@noreply.github.com>
> Datum: 03.09.2020 10:13
> Předmět: [gotthardp/lorawan-server] ** URGENT ** I cant restart LoRaserver, the console windows crashes LOG-files attached (#745)
 
Hi
 I cant restart Lorawan server 0.6.7, i tried to right click and restart .bat file as admin.. the console window crashes. Can anyone spot whats wrong from the attached log files ? Erlang is version OTP22.
debug.log <https://github.com/gotthardp/lorawan-server/files/5167123/debug.log>
 error.log <https://github.com/gotthardp/lorawan-server/files/5167126/error.log>
 crash.log <https://github.com/gotthardp/lorawan-server/files/5167127/crash.log>

hjojohan, 2020-09-03:
Unfortunalty we dont have a fresh backup as the backup script dont generate any files in windows servers. Can the database be saved and rebuilded using a tool ?  
Do you have the possibillity to check, you know the system best. , I can give you RDP access to our desktop and we will pay for your help of course.  We can continue using email johan@gpslogik.com

hjojohan, 2020-09-03:
I have used snapshoots from my virtual server to restore backup from 1 day, 3 day and one week old version of the whole server, still its not possbile to start the Lorawan server.