gotthardp/lorawan-server #746: LoRaWAN Server 0.6.7 crashes on start

ivangarrera, 2020-09-14:
Hello, 

I've been using the server with no problems for several months. However, it crashed yesterday and I cannot start it anymore. When I try to execute:
`sudo service lorawan-server start`
the server crashes. The output of the `crash.log`is:

`crasher:
    initial call: application_master:init/4
    pid: <0.289.0>
    registered_name: []
    exception exit: {{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{timeout,[gateway]}},[{lorawan_db,ensure_table,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_db.erl"},{line,117}]},{lists,foreach,2,[{file,"lists.erl"},{line,1338}]},{lorawan_app,start,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,18}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,277}]}]}}}},[{application_master,init,4,[{file,"application_master.erl"},{line,138}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
    ancestors: [<0.288.0>]
    message_queue_len: 1
    messages: [{'EXIT',<0.290.0>,normal}]
    links: [<0.288.0>,<0.43.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 1598
    stack_size: 27
    reductions: 265
  neighbours:`

Could anyone guess what is happening?

altishchenko, 2020-09-14:
@ivangarrera Your database got corrupted and cannot recover upon startup, especially the gateway table. Do you have recent backups of your database?

ivangarrera, 2020-09-14:
@altishchenko  No, I don't. Is there any way to recover the data? Would it be possible to know the possible reason why the database was corrupted?

altishchenko, 2020-09-14:
@ivangarrera As for the possible reason, this depends on your hardware, available disk space etc. If you don't have backups (which is not a good practice), you can try restoring _most_ of your data, but loosing the gateway table - you will need to re-enter your gateways configurations after all.
Now, what you can try:
1. **Make a copy of your database files** (should be in /var/lib/lorawan/, directory named Mnesia.xxxx@xxxx), so you can restore them if nothing helps.

```bash
sudo lorawan
cd /var/lib/lorawan/
erl -sname lorawan
Eshell V11.0.2  (abort with ^G)
(lorawan@....)1> mnesia:start().
```

Then try this and post results:
```
(lorawan@....)2> mnesia:wait_for_tables([gateway], infinity).
(lorawan@....)3> mnesia:force_load_table(gateway).
(lorawan@....)4> mnesia:stop().
```

Try running your server. If this won't help, you will need to delete the gateway table and let the server re-create it, you will loose your gateways configs though.

ivangarrera, 2020-09-14:
@altishchenko I've been able to restore a recent backup. Thank you very much!

closed 2020-09-14 by ivangarrera