gotthardp/lorawan-server #748: Debugging Erlang build function

nupik, 2020-09-21:
Hello! It will be very helpful if possible to try build function (Build Downlink) with online compiler. For example: https://rextester.com/l/erlang/
Could you provide example code for it, where I can enter JSON and see payload result. And some hint: if we have different JSON fields.

altishchenko, 2020-09-21:
@nupik Hello, build function checking is not available at the moment. Do you need an example of the build function in Erlang, so you can construct your own?

nupik, 2020-09-25:
Yes, I am not very familiar with erlang, what is why I need to test function before put it on server. Please could you provide example like for issues #625

altishchenko, 2020-09-30:
@nupik Sorry for the delay. Basically, build downlink takes one parameter - the map built from the fields of your JSON data and returns a binary object to be sent to the device. Opposed to the build uplink, which does the reverse - takes the binary and returns a map.

Here is a short example:

```json
{
    "mode": 2,
    "time": 1601505129,
    "delay": 3000,
    "port": 2,
    "devaddr": "01020304",
    ...
}
```

```
fun(#{command := Cmd}) ->
    << 1, Cmd >>;
(#{mode := Mode, time := Time, delay := Delay}) ->
    << 2, Mode, Time:32, Delay:16>>
end.
```

Two notes here:
- You are picking the fields from the map you are only interested in, there is no need to describe all possible fields.
- You can write multiple clauses to the fun, depending on the presence or absence of a particular JSON field.

If you can share an example of your JSON and the expected result, I can sketch the fun for you.

I strongly suggest, you have a quick look over the relevant chapters of [this book](https://learnyousomeerlang.com/content),
especially Anonymous functions.

nupik, 2020-10-06:
Hi, thanks in advance for your help. So we want able to send several commands. I try use fun(see below), but it not working and default Build Downlink is not work too. 
How can I use default request: ({"devaddr":"01020304","data":"710263","port":2,"time":"immediately"}) and custom in same time?

fun(#{mode := Mode}) ->
       << 16#1B:8, Mode, 16#00:8 >>;
    (#{set:= Set}) ->
        << 16#23:8,16#80:8,Set,16#00:8>>;
    (#{level:= Level}) ->
        << 16#1C:8,16#80:8,Level,16#00:8>>
end.

nupik, 2020-10-08:
Please, I need help!

altishchenko, 2020-10-12:
@nupik Hello, let's start from the beginning. What is the source JSON, you are trying to parse as custom? You can and should omit these `:8` postfixes, as this is the default. Also, in your list of funs I don't see a clause for the default behaviour. That is, the last fun should be:

```
(Map) ->
    Map
end.
```

nupik, 2020-10-20:
Custom JSON is {"devaddr":"01020304","mode":"01","port":2,"time":"immediately"} or {"devaddr":"01020304","set":"64","port":2,"time":"immediately"} or {"devaddr":"01020304","level":"40","port":2,"time":"immediately"} or default 
({"devaddr":"01020304","data":"710263","port":2,"time":"immediately"})
This is right? How I can check synaxis in https://rextester.com/l/erlang/ ? 
fun(#{mode := Mode}) ->
<< 16#1B, Mode, 16#00 >>;
(#{set:= Set}) ->
<< 16#23,16#80,Set,16#00>>;
(#{level:= Level}) ->
<< 16#1C,16#80,Level,16#00>>
(Map) ->
    Map
end.

altishchenko, 2020-10-20:
@nupik Semicolon is missing after the level clause. I don't know what you can do with rextester, to test your funs properly you will need access to the jsx library (included in the server).

```erlang
fun(#{mode := Mode}) ->
    << 16#1B, Mode, 16#00 >>;
(#{set:= Set}) ->
    << 16#23,16#80,Set,16#00>>;
(#{level:= Level}) ->
    << 16#1C,16#80,Level,16#00>>;    % <-- semicolon was missing here
(Map) ->
    Map
end.
```

nupik, 2020-10-22:
Not working :( . With rextester I want to test synaxis and result like in issue #625
[image]
[image]

altishchenko, 2020-10-22:
Aha! Sure it won't - I didn't pay attention to value *type* of the mode parameter. It is a string! If you are passing a string, you should do it like this:
```erlang
<< 16#1B, Mode/binary, 16#00 >>
```

If you want this presumably HEX being converted to a raw data before sending over, you should do like this:

```erlang
fun(#{mode := Mode}) ->
    Bytes = lorawan_utils:hex_to_binary(Mode),
    << 16#1B, Bytes/binary, 16#00 >>;
```