gotthardp/lorawan-server #750: tx_freq

swapmicro, 2020-10-01:
Hello There,
Recently I am trying to implement with sx1262 with US915 with the frequency of 914300000 with ABP mode at that time on the server I facing a tx_freq issue in the event log how can I solve this issue please guide me to solve this issue.
here I attaching my event window
[image]

Apaisal, 2020-10-02:
Please see network setting in infrastructure document and match them with your gateway frequencies setting at

https://github.com/gotthardp/lorawan-server/blob/master/doc/Infrastructure.md

Yours sincerely 
Anol P. <anol.p@emone.co.th>

swapmicro, 2020-10-03:
Hello there,
Thanks for the replay but I am facing the same error. 
Here I attached my Network configuration setting please guide me if I am wrong.
Thanks 
Swapnil 
[image]
[image]
[image]
[image]

altishchenko, 2020-10-03:
Hello, I am not sure how this works for US bands, but for EU you *must* list all the extra channels by hand on the channels tab. Default channels shall not be listed. I know, this inconvenient, but... Consider this:
[image]

etc. ...

swapmicro, 2020-10-05:
Hello There,
Thanks For the replay, I got the downlink and tx_err also solve by doing the settings in the network tab. Here I am posting my settings for US Frequency.
[image]
[image]
[image]

closed 2020-10-05 by swapmicro