gotthardp/lorawan-server #751: Down link Data queue issue

swapmicro, 2020-10-05:
Hello There,
When I trying to send multiple data frem to my single node at that time my node accepts the single frame. Other Frames are sending to the node as an empty frame here I am sending my node logs as well as server node queue
[image]
[image]
Please guide me to solve this problem.

altishchenko, 2020-10-05:
According to your screenshot, you can notice that the frames following your first received data frame are destined for port 0, which is an indication of ADR message exchange in action. Do you have ADR support switched on in your device's library? (LoRaMac-node, as I can guess). Bear in mind, ADR exchange may take considerable time. Does the **second queued frame** disappear from the server when the first one is successfully transmitted?

```
                /* Should be somewhere in your device's DEVICE_START section */
                mibReq.Type = MIB_ADR;
                mibReq.Param.AdrEnable = LORAWAN_ADR_ON;
                LoRaMacMibSetRequestConfirm(&mibReq);
```

swapmicro, 2020-10-06:
Hello There, 
Thanks for the replay.
Yes, After the receiving  1 frame. In semtec stack, code uses the **LORA_TxNeeded** function to get other pending frames.
[image]

swapmicro, 2020-10-06:
Hello There,
Yes In my device library ADR support is enabled.
If I have more then 1 frame is in the queue at that time my stack will use the  **LORA_TxNeeded** function to provide uplink as soon as possible at that time pending frames are sent from the server but I didn't get the data.
[image]

altishchenko, 2020-10-12:
@swapmicro Hello, sorry for the delay, I was unavailable. Does this call to TxNeeded() really happen? I am not sure at this moment, that this server will set this indication in the downlink even if it has some data pending. Also, this call may happen, if the library stack **has some ADR data** to be sent back to the server asap. Did you try it with the latest version of the server, top of master branch, just to confirm? You can get the docker image from docker hub to save the hassle on compile.

swapmicro, 2020-10-13:
Hello there,
Thanks for the reply.
Yes, when more than 1 frames are available in the activated (nodes) tab at that time **TxNeeded()**  called by the stack.
yes, I am using the latest copy of the server. 
Thanks 
Swapnil

altishchenko, 2020-10-13:
Hmm, ran out of ideas really... Can you possibly provide a fresh server debug.log with multiple downlink frames being queued/transmitted? Also, can you share an example of your downlink JSON?

swapmicro, 2020-10-14:
Hello There,
Thanks for the reply. Here  I am sharing my server fresh screen short.
[image]
[image]
[image]
[image]
Can you please guide me where can I found server debug,log file in rspbian.
my downlink json is 
{"devaddr":"0190A703","data":"4142434445464748494a4b4c4d4e4f505152535455565758595a313233343536373839306162636465666768696a6b6c6d6e6f707172737475767778797a"}
Thanks 
swapnil

altishchenko, 2020-10-15:
@swapmicro You can find your debug.log in /var/log/lorawan-server/.
For the downlink to reach the target device properly you should specify a `port` field in your JSON. Like this: `{ "port": 2, ... }`

swapmicro, 2020-10-16:
hello there,
Thanks for the reply.
adding  **"port":2** in my json my problem us solve.
Thanks 
Swapnil

closed 2020-10-16 by swapmicro