gotthardp/lorawan-server #753: Howto remove those 'unknown_deveui' from UI ??

fthiebolt, 2020-10-20:
Dear,

We're facing an increasing number of 'unknown_deveui' devices (Sigfox ?) ... is there any option to remove them from the UI ??

Merci bien
François

[image]

altishchenko, 2020-10-20:
Hello!
Did you try setting on your server Infrastructure -> Areas -> "Your area" -> 'Log ignored: false' ?

fthiebolt, 2020-10-21:
Dear Alexander

Thanks for your hint: in fact, it only stopped to show others LoRaWAN nodes (i.e the grey ones) ... but still appear the 'unknown_deveui' items !

We'll need to have a look to the code itself.

Merci
François