gotthardp/lorawan-server #754: Problems publishing on RabbitMQ and amqp protocol

scopelemanuele, 2020-10-25:
LoraServer connects correctly to the Brocker, but does not publish any data and does not generate any errors on the log:

2020-10-25 21: 26: 12.589 [debug] <0.308.0> @lorawan_connector_sup: stop_child: 23 Stop {amqp, << "RabbitRouter" >>}
2020-10-25 21: 26: 12.589 [debug] <0.11147.5> @lorawan_connector_amqp: log_termination: 160 Connector RabbitRouter terminated: shutdown
2020-10-25 21: 26: 12.592 [debug] <0.308.0> @lorawan_connector_sup: start_child: 16 Start {amqp, << "RabbitRouter" >>}
2020-10-25 21: 26: 12.593 [debug] <0.11177.5> @lorawan_connector_amqp: handle_info: 51 Connecting RabbitRouter to amqp: //mqtt.silvergeko.it/gateway
2020-10-25 21: 26: 12.593 [debug] <0.307.0> Supervisor lorawan_connector_sup started lorawan_connector_amqp: start_link ({connector, << "RabbitRouter" >>, << "Station" >>, << "json"> >, << "amqp: //mqtt.silvergeko.it/gateway" >>, ...}) at pid <0.11177.5>
2020-10-25 21: 26: 12.593 [debug] <0.11178.5> Supervisor {<0.11178.5>, amqp_connection_sup} started amqp_connection_type_sup: start_link () at pid <0.11179.5>
2020-10-25 21: 26: 12.593 [debug] <0.11178.5> Supervisor {<0.11178.5>, amqp_connection_sup} started amqp_gen_connection: start_link (<0.11179.5>, {amqp_params_network, << "gateway" >>, << "3Gauze.sulks" >>, << "gateway" >>, "mqtt.silvergeko.it", 5672,2047, ...}) at pid <0.11180.5>
2020-10-25 21: 26: 12.595 [debug] <0.11179.5> Supervisor {<0.11179.5>, amqp_connection_type_sup} started amqp_channel_sup_sup: start_link (network, <0.11180.5>, << "client 172.17.0.3:55313 -> 95.110.131.118:5672 ">>) at pid <0.11181.5>
2020-10-25 21: 26: 12.596 [debug] <0.11179.5> Supervisor {<0.11179.5>, amqp_connection_type_sup} started amqp_channels_manager: start_link (<0.11180.5>, << "client 172.17.0.3:55313 -> 95.110.131.118:5672 ">>, <0.11181.5>) at pid <0.11182.5>
2020-10-25 21: 26: 12.596 [debug] <0.11179.5> Supervisor {<0.11179.5>, amqp_connection_type_sup} started rabbit_writer: start_link (#Port <0.74084>, 0, 4096, rabbit_framing_amqp_0_9_1, <0.11180.5> , << "client 172.17.0.3:55313 -> 95.110.131.118:5672" >>) at pid <0.11183.5>
2020-10-25 21: 26: 12.596 [debug] <0.11179.5> Supervisor {<0.11179.5>, amqp_connection_type_sup} started amqp_main_reader: start_link (#Port <0.74084>, <0.11180.5>, <0.11182.5> , {method, rabbit_framing_amqp_0_9_1}, << "client 172.17.0.3:55313 -> 95.110.131.118:5672" >>) at pid <0.11184.5>
2020-10-25 21: 26: 12.597 [debug] <0.11179.5> Supervisor {<0.11179.5>, amqp_connection_type_sup} started rabbit_heartbeat: start_heartbeat_sender (#Port <0.74084>, 10, #Fun <amqp_network_connection.2.82346747>, { heartbeat_sender, << "client 172.17.0.3:55313 -> 95.110.131.118:5672" >>}) at pid <0.11185.5>
2020-10-25 21: 26: 12.598 [debug] <0.11179.5> Supervisor {<0.11179.5>, amqp_connection_type_sup} started rabbit_heartbeat: start_heartbeat_receiver (#Port <0.74084>, 10, #Fun <amqp_network_connection.3.82346747>, { heartbeat_receiver, << "client 172.17.0.3:55313 -> 95.110.131.118:5672" >>}) at pid <0.11186.5>
2020-10-25 21: 26: 12.598 [debug] <0.11187.5> Supervisor {<0.11187.5>, amqp_channel_sup} started amqp_gen_consumer: start_link (amqp_selective_consumer, [], {<< "client 172.17.0.3:55313 -> 95.110.131.118:5672">>,1}) at pid <0.11188.5>
2020-10-25 21: 26: 12.598 [debug] <0.11187.5> Supervisor {<0.11187.5>, amqp_channel_sup} started amqp_channel: start_link (network, <0.11180.5>, 1, <0.11188.5>, { << "client 172.17.0.3:55313 -> 95.110.131.118:5672">>,1}) at pid <0.11189.5>
2020-10-25 21: 26: 12.598 [debug] <0.11187.5> Supervisor {<0.11187.5>, amqp_channel_sup} started rabbit_writer: start_link (#Port <0.74084>, 1, 131072, rabbit_framing_amqp_0_9_1, <0.11189.5> , {<< "client 172.17.0.3:55313 -> 95.110.131.118:5672">>,1}) at pid <0.11190.5>
2020-10-25 21: 26: 12.599 [debug] <0.11177.5> @lorawan_connector_amqp: handle_subscribe: 175 Subscribing RabbitRouter to LoRaWan.gateway.apps.station.data.download

The publication queue is:
LoRaWan.gateway.apps.station.data.upload

While with the MQTT protocol it works correctly.

Thank you very much for helping

[image]

altishchenko, 2020-10-25:
Can you, please, show me how are trying to receive those values from AMQP? No errors in the log means that the message is really published to an amq.topic exchange with the relevant data. Also notice, that in the mqtt publish you use .../app/stazione/.. while in amqp you use .../apps/...

scopelemanuele, 2020-10-26:
The broker is RabbitMQ on docker image rabbitmq:3.8.9-alpine with enabled mqtt plugin.

This is the log on connection (LoRaWan Server):
2020-10-26 18:38:53.327 [info] <0.1503.0> accepting AMQP connection <0.1503.0> (172.17.0.1:40726 -> 172.17.0.4:5672)
2020-10-26 18:38:53.330 [info] <0.1503.0> connection <0.1503.0> (172.17.0.1:40726 -> 172.17.0.4:5672): user 'gateway' authenticated and granted access to vhost 'gateway'

This in the consumer python script:
```

import pika
import json
from EVOencoder import EvoTranscoder

def on_data(ch, method, properties, body):
    print(body)
    decoder = EvoTranscoder()
    payload = json.loads(body.payload.decode("utf-8", "ignore"))
    print(payload["text"])
    print(decoder.decode(payload["text"]))

credentials = pika.PlainCredentials(
    'gateway', 'password')
connection = pika.BlockingConnection(
    pika.ConnectionParameters('mqtt.silvergeko.it', credentials=credentials, virtual_host="gateway"))
channel = connection.channel()
channel.queue_declare(queue='LoRaWan.gateway.apps.stazione.data.upload')
channel.basic_consume(queue='LoRaWan.gateway.apps.stazione.data.upload',
                      auto_ack=True,
                      on_message_callback=on_data)
print(' [*] Waiting for messages. To exit press CTRL+C')
channel.start_consuming()
```

This is a log of a simple producer python script:
2020-10-26 18:41:53.161 [info] <0.1741.0> accepting AMQP connection <0.1741.0> (93.41.226.151:50998 -> 172.17.0.4:5672)
2020-10-26 18:41:53.206 [info] <0.1741.0> connection <0.1741.0> (93.41.226.151:50998 -> 172.17.0.4:5672): user 'gateway' authenticated and granted access to vhost 'gateway'
2020-10-26 18:41:53.316 [info] <0.1741.0> closing AMQP connection <0.1741.0> (93.41.226.151:50998 -> 172.17.0.4:5672, vhost: 'gateway', user: 'gateway')

This is the code of the producer script:
```

import pika

credentials = pika.PlainCredentials(
    'gateway', 'password')
connection = pika.BlockingConnection(
    pika.ConnectionParameters('mqtt.silvergeko.it', credentials=credentials, virtual_host="gateway"))
channel = connection.channel()
channel.queue_declare(queue='LoRaWan.gateway.apps.stazione.data.upload')

channel.basic_publish(
    exchange="", routing_key='LoRaWan.gateway.apps.stazione.data.upload', body='Hello World!')
print(" [x] Sent 'Hello World!'")
connection.close()
```
The message written with the producer script is successfully received by the consumer script.

altishchenko, 2020-10-26:
You missing **amq.topic** exchange binding for your clients.
As your producer lower uses *default* exchange, so does your consumer. You can run your producer and your consumer will get it. But server doesn't publish to *default* gateway, it publishes to **amq.topic** you need to bind your queue to get the data.

For your consumer. (You can check this by changing exchange="" to exchange="amq.topic" in your producer).

```python
result = channel.queue_declare(exclusive=True)
queue  = result.method.queue
channel.queue_bind(exchange="amq.topic", queue=queue)
```

scopelemanuele, 2020-10-27:
Resolved!!!! Thank you very much for your information!!!

closed 2020-10-27 by scopelemanuele