gotthardp/lorawan-server #756: Server crashes when encountering an email sending error

jcn, 2020-11-08:
I encountered an issue where our server would die after attempting to send a malformed email notification. I've attempted to address the specific issue in #755 but this probably shouldn't have caused the server to irrecoverably fail.

These is what showed up in `debug.log`:

```
2020-11-06 16:33:34.742 [warning] <0.396.0>@lorawan_db_guard:send_alert:196 gateway 0080000000010985 [{<<"disconnected">>,100}] []
2020-11-06 16:33:34.802 [warning] <0.396.0>@lorawan_db_guard:send_alert:196 gateway 0080000000016F12 [] [{<<"disconnected">>,100}]
2020-11-06 16:33:34.876 [error] emulator Error in process <0.407.0> on node lorawan@lorawan with exit value: {{nocatch,{permanent_failure,<<"554 Transaction failed: Illegal semicolon, not in group\r\n">>}},[{gen_smtp_client,try_DATA,4,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,333}]},{gen_smtp_client,send_it,2,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,193}]},{gen_smtp_client,send_it_nonblock,3,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,114}]}]}
2020-11-06 16:33:34.876 [debug] <0.336.0>@lorawan_connector_mqtt:terminate:259 Connector [redacted] terminated: shutdown
2020-11-06 16:33:34.876 [error] <0.320.0> Supervisor {<0.320.0>,lorawan_sup} had child db_guard started with lorawan_db_guard:start_link() at <0.396.0> exit with reason {nocatch,{permanent_failure,<<"554 Transaction failed: Illegal semicolon, not in group\r\n">>}} in gen_smtp_client:try_DATA/4 line 333 in context child_terminated
2020-11-06 16:33:34.877 [error] <0.320.0> Supervisor {<0.320.0>,lorawan_sup} had child db_guard started with lorawan_db_guard:start_link() at <0.396.0> exit with reason reached_max_restart_intensity in context shutdown
2020-11-06 16:33:34.877 [error] <0.412.0> CRASH REPORT Process <0.412.0> with 0 neighbours exited with reason: {owner_gone,{{nocatch,{permanent_failure,<<"554 Transaction failed: Illegal semicolon, not in group\r\n">>}},[{gen_smtp_client,try_DATA,4,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,333}]},{gen_smtp_client,send_it,2,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,193}]}, gen_smtp_client,send_it_nonblock,3,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,114}]}]}} in gun:owner_gone/1 line 966 in gun:proc_lib_hack/5 line 654
2020-11-06 16:33:34.877 [debug] <0.333.0>@lorawan_connector_mqtt:terminate:259 Connector Netvox terminated: shutdown
2020-11-06 16:33:34.877 [error] <0.267.0> Supervisor gun_sup had child gun started with {gun,start_link,undefined} at <0.412.0> exit with reason {owner_gone,{{nocatch,{permanent_failure,<<"554 Transaction failed: Illegal semicolon, not in group\r\n">>}},[{gen_smtp_client,try_DATA,4,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,333}]},{gen_smtp_client,send_it,2,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,193}]},{gen_smtp_client,send_it_nonblock,3,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,114}]}]}} in gun:owner_gone/1 line 966 in context child_terminated
2020-11-06 16:33:34.879 [info] <0.33.0> Application lorawan_server exited with reason: shutdown

```

I believe that first line is the gateway disconnect that the server was attempting to send the email warning about. After that, you can see that we got an error from `gen_smtp_client` in attempting to send the email, and then the server itself just dies, seemingly after a number of retries. I run `lorawan-server` in Docker and the Docker container just died in this case. When I would go to restart the container, it _would_ start up long enough for me to get a login prompt (and the logs show that the server process was getting restarted by the supervisor) but eventually, by the time I tried to log in, the process wold be dead again.

Interestingly, when I revoked the email user's credentials, I got an error, but the server process didn't die, so clearly it was getting handled better.

```
2020-11-06 17:45:01.623 [info] <0.315.0> gateway 00800000A000120B {connected,{{172,19,0,1},44868}}
2020-11-06 17:45:02.269 [error] emulator Error in process <0.373.0> on node lorawan@lorawan with exit value: {{nocatch,{permanent_failure,<<"535 Authentication Credentials Invalid\r\n">>}},[{gen_smtp_client,try_DATA,4,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,333}]},{gen_smtp_client,send_it,2,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,193}]},{gen_smtp_client,send_it_nonblock,3,[{file,"/lorawan-server/_build/default/lib/gen_smtp/src/gen_smtp_client.erl"},{line,114}]}]}
2020-11-06 17:45:02.271 [error] <0.310.0> Supervisor {<0.310.0>,lorawan_sup} had child db_guard started with lorawan_db_guard:start_link() at <0.312.0> exit with reason {nocatch,{permanent_failure,<<"535 Authentication Credentials Invalid\r\n">>}} in gen_smtp_client:try_DATA/4 line 333 in context child_terminated
2020-11-06 17:45:02.271 [debug] <0.310.0> Supervisor {<0.310.0>,lorawan_sup} started lorawan_db_guard:start_link() at pid <0.397.0>
```

altishchenko, 2020-11-09:
@jcn Ok, I am on it, but give me some time please.

altishchenko, 2020-11-09:
@jcn From the logs, in the second case db_guard is not retrying to send the notification by e-mail, so it is not failing again and again and max restart count is not reached, which keep the main supervisor lorawan_sup up and running.
In the first case max restart on db_guard process is reached and lorawan_sup kills itself unable to fix the problem. If lorawan_sup dies, server dies.
I am trying to figure out what gen_smtp does so dirty/special that breaks the supervision tree and also, somehow, interacts with gun's proc_lib_hack to supervision tree.
Unfortunately, the server itself is not responsible for this issue, but those two libraries mentioned above, that break the rules.