gotthardp/lorawan-server #760: Gateway communication bad right hand side

Kralg, 2020-12-02:
Hello,
I am testing a customized gateway. Some time after good communication, server starts throwing errors: 

[error] gen_statem <0.13935.0> in state uplink terminated with reason: no match of right hand value {aborted,{badarith,[{lists,sum,2,[{file,"lists.erl"},{line,291}]},{lorawan_mac_commands,average_qs0,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,66}]},{lorawan_mac_commands,average_qs,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_commands.erl"},{line,63}]},{lorawan_mac_commands,append_qs,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default..."},...]},...]}} in lorawan_mac_commands:handle_fopts/4 line 19

Can you advice on what the reason can be ?

altishchenko, 2020-12-03:
@Kralg Hello, can you share a full debug.log file from the start of the good communication and up to this problem?

This server tries to track device performance status by issuing periodic DevStatusReq LoRaWAN commands, their invocation is controlled by the `devstat_gap` configuration variable.
What I can see from the logs, is that at some moment in time the processing of the DevStatusAns response fails with incorrect value in the statistics measurements.
I suspect that one of your devices' database records contains incorrect data. You can get the complete record by issuing an HTTP request of the form: `http://yourserver:8080/api/nodes/XXXXXXXX`, where XXXXXXXX is the devAddr of your end-node, the one that sends DevStatusAns right before the error shows up in the logs. The field in the record to look at is `devstat`.
As a quick counter measure you can try deleting this end device from the server and creating a new record for it, so that all previous statistics for this device gets deleted.

pk-E32, 2020-12-09:
Thank you Alexander, yes the problem was with rssi and snr from gateway. They there nulls as gateway was not transmitting them. When it does transmit errors do not appear.

Kralg, 2020-12-09:
Thank you Alexander. The problem was because gateway was not transmitting in json the SNR and rssi values and the node had null values and this probably was causing the troubles. When gateway started transmitting them, errors do not appear.

closed 2020-12-09 by Kralg