gotthardp/lorawan-server #764: HTTP/REST endpoint documentation

TomlDev, 2021-03-11:
Hey together,
first of all, nice work! I was wondering where to find a proper (and full) documentation about the endpoints available when trying to communicate over HTTP/REST with the LoRaWAN server.
All I could find (and I also checked a lot of other issues) was the description of creating a [Connector](https://github.com/gotthardp/lorawan-server/blob/master/doc/Connectors.md#httprest).
Is there a possibility to see the details like one would have if it was built with Swagger? Or somewhere in the code?
I'm especially interested in the up-link possibilities. Thanks in advance!

TomlDev, 2021-03-23:
I kinly ask if @altishchenko or @gotthardp maybe could give me a hint, I really can't find this information somewhere...

mrerdem, 2021-04-29:
You may analyze the response of the below command to sort out the details. Please change "users" in the url with respective sections such as "areas" or "nodes" and the IP address with the one that belongs to the server.

`curl -i -X GET http://192.168.1.2:8080/api/users --user admin:admin`

altishchenko, 2021-04-30:
Hello!
Sorry for the long delay - didn't check the issues for a long while. @TomlDev, can you be a little bit more specific on what you are looking for?

There are two classes of messages - the first class is the messages sent by the server upon data receipt and events and the second class is the control messages exchanged with the server.

Details on the control messages can be found in the Administration.md document, the API endpoints at least, the particular fields map directly to the fields in the corresponding forms of the web interface and the fields marked as mandatory there must be present in the messages.

As for the data up-link messages, consider the Handlers.md document, most of the fields are detailed there. The message format in exchange is standard JSON.

Let me know, if you need specific examples.

Data exchanged is mapped one-to-one from the Erlang structs used in mnesia database into the JSON representation and back.