gotthardp/lorawan-server #767: Lora modulation polarization inversion option

scopelemanuele, 2021-03-27:
Hi,
I have the following problem, the final nodes have set the option ipol=false , but the downlink packets sent by the server have this setting to true.

RXTX ~ (TXPKT): [down] {"txpk": {"imme": false, "tmst": 3966554993, "freq": 433.0, "rfch": 0, "powe": 20, "modu": " LORA "," datr ":" SF7BW125 "," codr ":" 4/5 ",**" ipol ": true,**" size ": 13," data ":" YCnECiQhgAAG6Sq8Eg == "}}

My devices with this configuration are not receiving the ack or command ...

By manually sending the downlink to the gateway without "ipol" or with the "ipol" set to false the node receives the ack or the command.

How can this field be changed?

Thank's

altishchenko, 2021-05-09:
@scopelemanuele Emanuele, is there any reason why your end devices **expect** the downlink signal the same polarity as uplink? By LoRaWAN physical interface definition all *downlink* frames must be sent with inversed polarity, so that in a multi-node/multi-gateway setup uplinks will not interfere and cause collisions with downlinks on the same channel/band/frequency.

Unfortunately, the only way to change this behaviour is to modify the server source code... file lorawan_gw_forwarder.erl, line 224...

altishchenko, 2021-06-03:
@scopelemanuele Emanuele, can you please close this issue or do you request an ability to set downlink polarization from the user interface? Be advised, changing downlink polarization violates LoRaWAN standard.

scopelemanuele, 2021-06-04:
Sorry, I've been very busy and haven't had time to respond. I contacted the vendor of the device and asked to fix the bug.

Thanks so much

closed 2021-06-04 by scopelemanuele