gotthardp/lorawan-server #773: Question: Battery as % in uplink

glennvdv, 2021-05-25:
I'm trying to send the battery as % through the parse uplink function. But i can't seem to get it working, how do i access battery in the erlang function? My current function is:
fun(Fields,<<Header,Status,Casing_Status,Unused,DQ_Enabled,Counting:16,Temp:16>>)->
Fields#{status => Status, casing_status => Casing_Status,counting=>Counting,temperature=>Temp}
end.
I would like to send batteryLevel => battery / 255 *100

altishchenko, 2021-05-25:
For this, your Handler's Uplink fields should contain the word 'battery', then you can access the LoRa value from inside your function with:
```
LoRaBattery = maps:get(battery, Fields, 0),
```

Bear in mind, that dividing by 255 is not really correct. According to the standard, 0 - connected to external power supply, 255 - unable to measure. That leaves you with [1-254] = [0-100]%

glennvdv, 2021-05-25:
thx it's working. And thx for mentioning the standard, i did not know about the 0 and 255

closed 2021-05-25 by glennvdv