gotthardp/lorawan-server #774: missing "erlang-tools" in installation docs

Enemoino, 2021-05-28:
Hi i have branch of your code from 2019 right before commit "78c02ac27eaeb652b5abd20b38da09c6a7f8dc91",  and i was compiling version 0.7.0 from it and i had zero problems. Now i was looking for help and tried to compile new versions from @altishchenko and i have failed because now you have to install "erlang-tools" to "make release". 

Please add this to you doc/installation, in required erlang modules. 
Thank you

altishchenko, 2021-05-28:
Also, have a look at commit 9fe484422a58cc3a250ea72c8fbee6eade5d89d1, could this be the reason? `erlang-tools` could possibly be a requirement of one of the updated project dependencies and this sort of things doesn't show up in docker builds.

Enemoino, 2021-05-28:
Yes i went trought that commit as first suspect, i dint find any deps for erlang-tools but they could they be deeper than rebar.config ?

altishchenko, 2021-05-28:
> Yes i went trought that commit as first suspect, i dint find any deps for erlang-tools but they could they be deeper than rebar.config ?

Surely, that is what I meant. New version of a package may require a new version of another package and that may have introduced profiling or other stage into its build/install sequence... You never know, updating dependencies is always like a mine field.