gotthardp/lorawan-server #775: Connector error

msu1ao, 2021-06-01:
Hello
I installed the server on two different Win10 computers. I can create a handler, but as soon as I create a connector with Websocket I get errors. I can no longer edit the connector, only delete it. When I activate the connector, the server crashes.

Can someone help me there?

[crash.log](https://github.com/gotthardp/lorawan-server/files/6575864/crash.log)
[debug.log](https://github.com/gotthardp/lorawan-server/files/6575865/debug.log)
[error.log](https://github.com/gotthardp/lorawan-server/files/6575866/error.log)

altishchenko, 2021-06-01:
@msu1ao Please, post here your connector configuration. Screenshot will do.

msu1ao, 2021-06-02:
Thank you very much
Here is the screenshot, I didn't activate the connector, otherwise the server crashes.

[image]

altishchenko, 2021-06-02:
@msu1ao  OUCH! I didn't notice that in the first place!! You are using Erlang/OTP 24, which is **NOT** supported - lots of functions got deprecated and removed from the Erlang library not all packages that this server uses upgraded to 24 either. You must use Erlang 23, 22 or 21! Your best option is to run this server in the docker container for the moment - there is a pre-build image available on docker hub and instructions on how to deploy one properly.

msu1ao, 2021-06-03:
Hello

That was the problem, now it works. I'll take a look at Docker. Thanks for the help.

closed 2021-06-03 by msu1ao