gotthardp/lorawan-server #777: Unable to receive Downlink Msgs from server

pshingala145, 2021-06-05:
I m using  0.6.7 server and below is configuration parameters on ADR, i m sending CNF msgs from device to server and can see log on server, though server is not able to generate response to the CNF msg, i cannot see port on downlink msg which are generated by server. could you please suggest some solution on this. 

i have attached the server LOG and configuration parameters on LoRa US RN2903A .

const RADIO_MODE = "lora";
const RADIO_FREQ = 903300000;
const RADIO_SPREADING_FACTOR = "sf12"; // 128 chips 
const RADIO_BANDWIDTH = 250;           //500
const RADIO_CODING_RATE = "4/5";
const RADIO_CRC = "off"; // CRC header enabled
const RADIO_SYNC_WORD = 12;
const RADIO_WATCHDOG_TIMEOUT = 0;
const RADIO_POWER_OUT = 14;
"mac set rxdelay1 1000",
"mac set rx2 11 923300000",

please let me know if u need further details regarding same.

[image]

[image]

[Selection_003](https://user-images.githubusercontent.com/85383974/120881944-4d9fae00-c5f2-11eb-849e-9e39e9cb0e92.png)

altishchenko, 2021-06-05:
RX1 setting on your device is 1 second, but you set 2 seconds on your server, strange enough that your device receives any downlinks at all. What you see in the Frames log are acknowledgements sent by the server upon receipt of your uplink messages. If there is some other specific data should go downlink, you should verify that data is getting to the server correctly. How do you send downlinks? What is your Handler/Connector configuration? Do you *yourself* specify port number in your downlinks? Because you must.

pshingala145, 2021-06-08:
Thank you for your input!

I am able to receive downlink msgs.

altishchenko, 2021-06-08:
That is great! If the problem is solved, can you please close this issue?

closed 2021-06-08 by pshingala145