gotthardp/lorawan-server #781: Using {port} in the Publish Uplinks field of MQTT Connector

edulbastos, 2021-06-17:
Hi,

When using {port} in the Publish Uplinks field of MQTT Connector, the {port} is not translated to the specific port (the port parameter is in the uplink field) and I don´t receive any message in the MQTT server. I want to separate different ports to differents topics in the MQTT servers. Any hints?

Regards,

Eduardo Bastos

altishchenko, 2021-07-03:
Hello, can you show me how are you using this parameter? Passing port field by default (in the Fields parameter) works fine.

beavis88, 2021-12-21:
I have the same issue. I'm not receiving any MQTT topics when publish uplinks  parameter includes {port}. All works fine when {port} is removed.
My handlers "Parse Uplink" is default (empty).

altishchenko, 2021-12-31:
Hello, can you share your handler and connector configuration, so I may have a look on what is going wrong?

edulbastos, 2022-01-03:
Hi @altishchenko,

Sorry about the delay. I´m using {port} after the published topic. Ex: (in the Publish Uplinks field):

TTGO/{port}

But I´m not receiving any MQTT topic with port attached at the end. The attribute port is in the Uplink Fields. 

Regards,

Eduardo Bastos

edulbastos, 2022-01-03:
Hi @altishchenko,

The screenshots of my Connector and Handler configuration are as follow. 

[image]
[image]

Regards,

Eduardo Bastos

edulbastos, 2022-01-03:
Hi @altishchenko,

The screenshots of my Handler and Connector configuration are as follow:

[image]
[image]

Regards,

Eduardo Bastos

beavis88, 2022-01-10:
My settings for handler:
			downlink_expires = "never";
			event_fields = ("app", "event", "appid", "devaddr", "appargs", "datetime", "deveui", "port");
			payload = "custom";
			uplink_fields = new JArray("port", "app", "data", "datetime", "deveui", "desc", "freq", "codr", "datr", "fcnt");
Connector:
			auth = "normal";
			enabled = true;
			format = "raw";
			publish_qos = 0;
			publish_uplinks = @"{app}/{port}";
			publish_events = @"{app}/events";
			received = @"{deveui}/tx";
			subscribe = "+/tx";
			subscribe_qos = 0;
			uri = "mqtt://127.0.0.1:1883";

All other fields are empty (default).