gotthardp/lorawan-server #783: If container is run in azure it creates a new database with each startup

compunited, 2021-06-24:
this happens after each start:
2021-06-24 10:14:16.533 [debug] <0.121.0>@lager_handler_watcher:127 Lager installed handler {lager_file_backend,"error.log"} into lager_event
2021-06-24 10:14:16.536 [debug] <0.127.0>@lager_handler_watcher:127 Lager installed handler error_logger_lager_h into error_logger
2021-06-24 10:14:16.998 [debug] <0.116.0>@lager_handler_watcher:127 Lager installed handler lager_backend_throttle into lager_event
2021-06-24 10:14:21.341 [debug] <0.233.0>@lorawan_app:start:17 Using config: [{map_tile_server,<<"http://{s}.tile.openstreetmap.org/{z}/{x}/{y}.png">>},{packet_forwarder_listen,[{port,1680}]},{slack_server,{"slack.com",443}},{deduplication_delay,200},{devstat_gap,{432000,96}},{http_admin_credentials,{<<"admin">>,<<"admin">>}},{max_lost_after_reset,10},{http_extra_headers,#{}},{trim_interval,3600},{http_admin_path,<<"/admin">>},{http_admin_listen_ssl,[{port,8443},{certfile,"cert.pem"},{cacertfile,"cacert.pem"},{keyfile,"key.pem"}]},{applications,[{<<"semtech-mote">>,lorawan_application_semtech_mote}]},{gateway_delay,200},{server_stats_interval,60},{connector_monitor_period,10000},{http_admin_listen,[{port,8080}]},{http_custom_web,[]},{ssl_options,[]},{connectors,[{lorawan_connector_amqp,[<<"amqp">>,<<"amqps">>]},{lorawan_connector_mqtt,[<<"mqtt">>,<<"mqtts">>]},{lorawan_connector_http,[<<"http">>,<<"https">>]},{lorawan_connector_mongodb,[<<"mongodb">>]},{lorawan_connector_ws,[<<"ws">>]}]},{frames_before_adr,50},{event_lifetime,86400},{retained_rxframes,50},{websocket_timeout,3600000},{http_admin_redirect_ssl,true}]
2021-06-24 10:14:21.342 [info] <0.43.0> Application mnesia exited with reason: stopped
2021-06-24 10:14:21.342 [info] <0.233.0>@lorawan_db:ensure_tables:25 Database create schema
2021-06-24 10:14:21.924 [info] <0.233.0>@lorawan_db:create_table:148 Database create config
2021-06-24 10:14:22.122 [info] <0.233.0>@lorawan_db:create_table:148 Database create user
2021-06-24 10:14:22.402 [info] <0.233.0>@lorawan_db:set_defaults:263 Database create default user:password

here you can see the result
[image]

compunited, 2021-06-25:
since it's currently impossible to change the hostname in azure it's not the best idea to use Mnesia.lorawan@hostname and make it not overwritable 
https://feedback.azure.com/forums/602224-azure-container-instances/suggestions/42509800-support-setting-of-hostname

altishchenko, 2021-07-01:
Mnesia is a builtin Erlang/OTP database and it is not the server's to decide how it stores its files. Changing fqdn in erlang/otp is not a trivial task, if at all possible.