gotthardp/lorawan-server #792: Compressed Payload

aogonzalez75, 2021-08-30:
Dear All, we are trying to compress the payload before send and then decompress before publish via MQTT? I will appreciate your support. Thanks

altishchenko, 2021-08-30:
Hello, I am not sure that I understood you correctly, but is your question "How to do something with received payload, decompress for example, before publishing it over MQTT?"
If so, take a look into this document: [Handlers.md](https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md#parse-uplink)

aogonzalez75, 2021-08-30:
for i.e, device compress 200 bytes with LZ4 algorithm to send lower size paylod. You received a binary compressed on the Handler. How to decompress it before publish?

altishchenko, 2021-09-01:
As I said before, read Handlers.md document. You have to write your custom Parse Uplink function if you want to do something with your received data before publishing it to any connector.