gotthardp/lorawan-server #793: Crash on new connector (mqtt or mongo)

jackkum, 2021-09-03:
There was a need to move the container to another server. I deployed the container on a new server and using api transferred data from one server to another. And after a couple of seconds, the application crashed. In attempts to find out the reason, I came to the conclusion that after creating a connector (any other than a web socket), a crash occurs.

Perhaps a solution to the problem is somewhere in sight, but I cannot find it.

[crash.log](https://github.com/gotthardp/lorawan-server/files/7103787/crash.log)
[error.log](https://github.com/gotthardp/lorawan-server/files/7103789/error.log)

altishchenko, 2021-09-03:
For whatever reason your mqtt connector fails on tlsv1.2 negotiation and, because it restarts automatically too often and too many times, it brings the system down, which it shouldn't.

altishchenko, 2021-09-03:
Did you copy certificates along?

jackkum, 2021-09-08:
@altishchenko Thank you!

closed 2021-09-08 by jackkum