gotthardp/lorawan-server #795: Main Page no load

dencorp63, 2021-09-09:
Realese 0.7.0
[image]
When loading the main page after 2 minutes, a blank screen.

When loading frames without a filter (by clicking on the Frames label), also after 1 minute
[image]

altishchenko, 2021-09-09:
Please provide error.log or debug.log file from your system.

dencorp63, 2021-09-10:
[debug.log](https://github.com/gotthardp/lorawan-server/files/7141967/debug.log)
Here is a part of the debug log at the moment we click on loading frames and the main window

altishchenko, 2021-09-10:
There are absolutely no related errors in the debug.log, are you sure that the server crashes? Do you have anything written in the crash.log possibly?

To make a proper experiment:

1. stop the server
2. erase the log files
3. start the server
4. try to reproduce the situation
5. make sure the server **crashed**
6. collect the logs and send them to me.

DO NOT cut the resulting log files.

Also, as a side note, make sure the server does not display any memory or disk related warnings near its name on the main dashboard.

dencorp63, 2021-09-13:
[log.zip](https://github.com/gotthardp/lorawan-server/files/7152013/log.zip)
Complete all steps. Logs in the attachment. The situation repeated itself.

altishchenko, 2021-09-13:
Ok, for what I can gather from the logs is that mnesia indexes are in inconsistent state - some saved references are non-existent in the data tables, which breaks ram caching. Quite possibly this is a result of a single crash long ago or a result of a db file being deleted or otherwise modified **by hand** or the file got corrupted under the low disk condition stress.

There is only one way out of this situation, if is at all possible in your DB condition:

1. Preferrably (! but you may risk without this step) cut the outside connections on a system level or unplug the ethernet cable, so nothing will get through while you are attempting to restore.
2. Export the server database with dbexport script provided in the server's bin directory (you may have to modify credentials in it, if you deleted admin/admin).
3. Stop the server.
3. Completely delete the mnesia database folder (or move it somewhere far away, in case it may become needed.)
4. Start the server: it will re-initialize the new database properly.
5. Import saved in step 2 configuration with dbimport script. **NOTE**: system users are exported without passwords! Make sure you don't have an 'admin' user in this file, this guarantees you will have an admin/admin access to the system after import. Users passwords will need to be reconfigured by hand or you can edit the users.json file adding a "pass" field to each user with clear text password in it.

If everything works ok, try to monitor the disk state for low disk condition and never ever touch any mnesia file by hand.

altishchenko, 2021-11-04:
Hi! Is the problem still there? Or, can you close the issue, please?