gotthardp/lorawan-server #796: Trouble with Connector MQTT after upgrading to Mosquitto 2.0.12 (from 2.0.11)

edulbastos, 2021-09-10:
Hi,

After upgrading my Debian/Buster mosquitto environment to 2.0.12 (from 2.0.11), my MQTT (mqtt://) connector stopped working. 

1631299480: Bad socket read/write on client 12345: Invalid arguments provided.

Anyone with this issue?

Regards,

Eduardo Bastos

edulbastos, 2021-09-11:
Hi,

Downgrading to mosquitto 2.0.11 solved my problem, using the SAME mosquitto.conf file. 

allow_anonymous false
password_file /etc/mosquitto/pwfile
listener 1883 XX.XX.XX.XX
listener 1883 localhost
listener 9001
protocol websockets
user mosquitto

It looks like it´s necessary to insert some configuration parameter in the new version of mosquitto to tweak some security issue. Any hints?

Regards,

Eduardo Bastos

spynappels, 2022-01-14:
I have the same issue, unfortunately I can not easily downgrade to mosquitto 2.0.11.
lorawan-server version is latest release, so I think that is 0.6

I am not sure what changed but I'd very much appreciate it if anyone had any idea on what could be changed, either on the lorawan-server side or the mosquitto side to make this work.

Has anyone been able to work out what the actual problem is, what parameter mosquitto doesn't like? 

I've not got any debug logging from Mosquitto, but this is what I do get:
```
1642167085: New connection from 127.0.0.1:46863 on port 1883.
1642167085: New client connected from 127.0.0.1:46863 as spttest (p1, c1, k0, u'spt-test').
1642167085: Bad socket read/write on client spttest: Invalid arguments provided.
```
I have also seen the same but with p2 instead of p1.
I know using mosquitto_sub I get the following, so could the keepalive parameter (`k0`) maybe be the problem?
```
1642166841: New connection from 127.0.0.1:43796 on port 1883.
1642166841: New client connected from 127.0.0.1:43796 as auto-DE45AA79-F23D-DB14-F2A6-ABF0EE559547 (p2, c1, k60, u'spt-test').
```

Albert611, 2022-03-11:
Hello, is there any tutorial on connecting lora-server to an external MQTT server?

lcse66, 2022-08-18:
Hello,
just upgrade to mosquitto >= 2.0.13 and add to mosquitto.conf:
`max_keepalive 0`

From 2.0.12 mosquitto refuses connections when client requests zero (no) keepalive.
From 2.0.13 mosquitto accepts that connections only if max_keepalive is set to zero.

```
2.0.12 - 2021-08-31
===================

Security:
[...]
- Fix `max_keepalive` not applying to MQTT v3.1.1 and v3.1 connections.
  These clients are now rejected if their keepalive value exceeds
  max_keepalive. This option allows CVE-2020-13849, which is for the MQTT
  v3.1.1 protocol itself rather than an implementation, to be addressed.
[...]
```
```
2.0.13 - 2021-10-27
===================

Broker:
- Fix `max_keepalive` option not being able to be set to 0.`
[...]
```

From [mosquitto changelog](https://mosquitto.org/ChangeLog.txt)

edulbastos, 2022-08-18:
Hi lcse66,

Thanks for the tip of "max_keepalive 0". It worked like a charm.

Regards,

Eduardo Bastos