gotthardp/lorawan-server #797: Binary comprehensions on Handler

aogonzalez75, 2021-09-11:
Dear All, I'm trying to convert received binary data to 16 bits integer list...

fun(Fields, <<Count:8, Data/binary>>) ->
  Regs=[ H ||<<H:16>>  <= Data ],
  Fields#{cnt=>Count, data=>Regs}
end.

altishchenko, 2021-09-13:
And your problem is?

aogonzalez75, 2021-09-13:
not working... sorry, I'm new with Erlang language, I'm not sure it is the correct way to  generate a field map from a list?

altishchenko, 2021-09-13:
Your function works ok.
I am not sure, if it is an expected behaviour, but your Regs variable becomes a List of the form [ 1, 3 ,4 ,5 ] etc.
What is not working really?

aogonzalez75, 2021-09-13:
no. In my understanding "Fields#" is a map... Im not sure it is possible assign a list to a map field (data=>Regs)?

altishchenko, 2021-09-14:
Yes, Fields is a map. And you are modifying certain keys in it. What your function does is:

1. It adds a new key to the map named `cnt` with a value of the byte `Count`.
2. Replaces the value of the key `data` with the value of the list `Regs`.

So, suppose on the entrance you `Fields` map looks like this:
```
Fields#{ devaddr => <<1, 2, 3, 4>>, data => <<3, 9, 8, 7, 6>>}
```

Upon exit your map will be modified into this:

```
Fields#{ devaddr => <<1,2,3,4>>, cnt => 3, data => [2057, 1543] }
```

aogonzalez75, 2021-09-15:
fun(Fields, <<Count:8, Data/binary>>) ->
  Regs=[ H ||<<H:16>> <= Data ],
  Fields#{cnt=>Count, data=>Regs}
end.

I attached log files...

[crash.log](https://github.com/gotthardp/lorawan-server/files/7169675/crash.log)
[debug.log](https://github.com/gotthardp/lorawan-server/files/7169676/debug.log)
[error.log](https://github.com/gotthardp/lorawan-server/files/7169677/error.log)

aogonzalez75, 2021-09-15:
fun(Fields, <<Count:8, Data/binary>>) ->
  Regs=[ H ||<<H:16>> <= Data ],
  Addr=lists:seq(1,Count),
  Fields#{addr=>Addr, cnt=>Count, data=>Regs}
end.

I attached log files...
[crash.log](https://github.com/gotthardp/lorawan-server/files/7169796/crash.log)
[debug.log](https://github.com/gotthardp/lorawan-server/files/7169797/debug.log)
[error.log](https://github.com/gotthardp/lorawan-server/files/7169798/error.log)

aogonzalez75, 2021-09-15:
solved...

fun(Fields, <<Count:8, Data/binary>>) ->
  Cnt=trunc(Count/2),
  Addr=lists:seq(1, Cnt),
  Regs=[ H ||<<H:16/little-integer>> <= Data ],
  Fields#{cnt=>Cnt, addr=>Addr, regs=>Regs}
end.

aogonzalez75, 2021-09-15:
Thanks a lot...

closed 2021-09-15 by aogonzalez75

altishchenko, 2021-09-15:
Oh! Your real solution was assigning Regs to `regs`, not to `data`. That is what the failure was - `data` is automatically converted to hex string on transmission.

aogonzalez75, 2021-09-15:
good to know... thanks