gotthardp/lorawan-server #798: HTTP Connection Issue

saini-jitendra, 2021-09-13:
Hello,

I want to send data to the http server, for which we have created Connector and Handler.  We have taken the reference of **connector. md** and **handler.md** documents.

Moreover we are using Basic authentication as its HTTP protocol. Unfortunately, data has not been received on HTTP Server. 

When we follow same procedure(URL, Basic authentication, payload) on POSTMAN, data has been received successfully. So, I suspect, there is not an issue with Http server.   

Below is the configuration of the connector.

Any Leads would be appreciated! 
**[image]**

[image]

[image]

altishchenko, 2021-09-13:
You should specify your `http://<something-top-secret>:5000` as a URI and your `sensors/api/v2/data` as a Publish Uplinks topic.

saini-jitendra, 2021-09-13:
@altishchenko 

I have tried to add  **/sensors/api/v2/data** in Publish uplink topic.  I have got HTTP error on the web page and data has not been received on the HTTP client-server.

Below is the configuration of the connector and web page logs.

[image]
[image]

altishchenko, 2021-09-13:
Can you have a look into the logs, to see what is the exact error?

Different IoT platforms require different extraordinary fields to be filled for the request to be accepted. There are certain hooks for this in the server, but in order to tell you which one to use you'll have to tell me what platform you are using or this platforms HTTP POST requirements. Unfortunately the word POSTMAN tells me nothing.

You must make this check at least: What happens if you post your  data to the platform server **by hand** (using curl for example)? What error do you get? Does the post go through ok?

And please, remove all these useless 'At most once' QoS settings - they are for MQTT ONLY.

saini-jitendra, 2021-09-13:
2021-09-13 09:26:39.309 [debug] <0.7365.27>@lorawan_connector_http:ensure_gun:160 Connecting NanoSense_HTTP_test to http://yyy.com:5000
2021-09-13 09:27:01.130 [debug] <0.7365.27>@lorawan_connector_http:handle_info:117 HTTP request failed: 404, {<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}
2021-09-13 09:27:01.130 [warning] <0.7365.27> connector_http {http_error,{404,<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}}
2021-09-13 09:27:01.328 [debug] <0.7421.27>@lorawan_mac_commands:build_fopts:77 69E62DC2 <- [dev_status_req]
2021-09-13 09:27:27.852 [warning] <0.354.0> gateway A840412042544150 {ack_lost,100.0}
2021-09-13 09:27:37.379 [debug] <0.7365.27>@lorawan_connector_http:handle_info:117 HTTP request failed: 404, {<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}
2021-09-13 09:27:37.379 [warning] <0.7365.27> connector_http {http_error,{404,<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}}
2021-09-13 09:27:37.577 [debug] <0.7458.27>@lorawan_mac_commands:build_fopts:77 69E62DC2 <- [dev_status_req]
2021-09-13 09:27:57.856 [warning] <0.354.0> gateway A840412042544150 {ack_lost,100.0}
2021-09-13 09:28:13.733 [debug] <0.7365.27>@lorawan_connector_http:handle_info:117 HTTP request failed: 404, {<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}
2021-09-13 09:28:13.733 [warning] <0.7365.27> connector_http {http_error,{404,<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}}
2021-09-13 09:28:13.929 [debug] <0.7507.27>@lorawan_mac_commands:build_fopts:77 69E62DC2 <- [dev_status_req]
2021-09-13 09:28:27.859 [warning] <0.354.0> gateway A840412042544150 {ack_lost,100.0}
2021-09-13 09:28:50.084 [debug] <0.7365.27>@lorawan_connector_http:handle_info:117 HTTP request failed: 404, {<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}
2021-09-13 09:28:50.084 [warning] <0.7365.27> connector_http {http_error,{404,<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}}
2021-09-13 09:28:50.282 [debug] <0.7532.27>@lorawan_mac_commands:build_fopts:77 69E62DC2 <- [dev_status_req]
2021-09-13 09:28:57.863 [warning] <0.354.0> gateway A840412042544150 {ack_lost,100.0}
2021-09-13 09:29:26.437 [debug] <0.7365.27>@lorawan_connector_http:handle_info:117 HTTP request failed: 404, {<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}
2021-09-13 09:29:26.438 [warning] <0.7365.27> connector_http {http_error,{404,<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}}
2021-09-13 09:29:26.634 [debug] <0.7565.27>@lorawan_mac_commands:build_fopts:77 69E62DC2 <- [dev_status_req]
2021-09-13 09:29:27.866 [warning] <0.354.0> gateway A840412042544150 {ack_lost,100.0}
2021-09-13 09:30:02.788 [debug] <0.7365.27>@lorawan_connector_http:handle_info:117 HTTP request failed: 404, {<<"http://yyy.com:5000">>,<<"//sensors/api/v2/data">>}
2021-09-13 09:30:02.788 [warning] <0.7365.27> connector_http {http_error,{404,<<"http://dev.nanoprecisedataservices.com:5000">>,<<"//sensors/api/v2/data">>}}
2021-09-13 09:30:02.986 [debug] <0.7588.27>@lorawan_mac_commands:build_fopts:77 69E62DC2 <- [dev_status_req]
2021-09-13 09:30:27.975 [warning] <0.354.0> gateway A840412042544150 {ack_lost,100.0}
2021-09-13 09:30:39.038 [debug] <0.7365.27>@lorawan_connector_http:handle_info:117 HTTP request failed: 404, {<<"http://yyy:5000">>,<<"//sensors/api/v2/data">>}

Above is the LNS Logs.

altishchenko, 2021-09-13:
404, wrong URL. So, what happens if you try this with curl from command line? Do you get 200 reply?

saini-jitendra, 2021-09-17:
Hello @altishchenko,

Thank You for your support!

The issue has been resolved by adding **/sensors/api/v2/data/** as a Publish Uplinks topic.

closed 2021-09-17 by saini-jitendra