gotthardp/lorawan-server #800: Unknown packet send by server to gateway/device w/o port number

DiezelMax, 2021-09-27:
Hello,
I'm coding/debugging gateway and when new device is first time connecting to server it is getting interesting payload.
This packet comes from server to gateway:
027D8E037B227478706B223A7B22706F7765223A31342C22696D6D65223A66616C73652C22746D7374223A313030303030302C22636F6472223A22342F35222C2264617472223A225346394257313235222C226D6F6475223A224C4F5241222C2272666368223A22756E646566696E6564222C2269706F6C223A747275652C2273697A65223A31332C2264617461223A22594659414141434241514147536A763442513D3D227D7D
Which contains JSON: {"txpk":{"powe":14,"imme":false,"tmst":1000000,"codr":"4/5","datr":"SF9BW125","modu":"LORA","rfch":"undefined","ipol":true,"size":13,"data":"YFYAAACBAQAGSjv4BQ=="}}
Payload for node is: "YFYAAACBAQAGSjv4BQ==", in HEX: 6056000000810100064a3bf805.
How to understand this frame? It is MAC commands?

altishchenko, 2021-09-29:
Yes, it is a DevStatusReq.
You can enable or disable device status query in the device's Profile, tab ADR, field 'Request Status'.

altishchenko, 2021-11-04:
Hi! If the question is answered, can you close the issue, please?

closed 2021-12-03 by DiezelMax