gotthardp/lorawan-server #802: OTAA join procedure

TheStudent227, 2021-10-03:
Hello, I would like to ask for help with lorawan network server. I am doing the project on CoDeSys for controller, which should emulate the network server. The end device is premade, but I have troubles finding the network server code which responsible for OTAA join procedure, which will handle the request from LoRa sensor. Could someone point on this code in this project ? Or have some example ?

Apologizing for possible off-topic

altishchenko, 2021-10-03:
If you need to figure out the logic, you can start with src/lorawan_mac.erl file, line 33. The word 'Join' will guide you through the process. There is another network server, called chirpstack, which is also available on GitHub.

The LoRaWAN standard is very straightforward, you can get the specification at [Lora-Alliance](https://lora-alliance.org/resource_hub/lorawan-104-specification-package/). There is no black magic in the protocol and it is very simple to code.

TheStudent227, 2021-10-03:
> If you need to figure out the logic, you can start with src/lorawan_mac.erl file, line 33. The word 'Join' will guide you through the process. There is another network server, called chirpstack, [...]

Thank you for help :)

altishchenko, 2021-11-04:
Hi! Can you close the issue, please?

closed 2021-11-16 by gotthardp