gotthardp/lorawan-server #803: How should I set the confirmed downlink message?

Ryoyo123, 2021-10-06:
Hi
 I want to send confirmed data via downlink to make sure ACK is returned. 

 The End node has installed　STMicro AT-Slave program . The gateway is in packet forwarder mode. 

 The uplink data path is end node-> packet forwarder GW-> LNS (gotthardo lorawan-server)-> MQTT broker-> Node-RED. 

>  Is there an API that sends confirmed data from LNS to end node by downlink? 
>  Could you advise me how to set it?

altishchenko, 2021-10-06:
There is a special section in [Handlers](https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md) concerning downlinks, confirmed downlinks, confirmed downlinks with receipt.
Read the docs first, please!

Ryoyo123, 2021-10-07:
Hi,
Thank you for your advice.

I read "Backend Handlers". 
Should I set "D / L Expires" to "Never" to set the downlink packet to confirmed data?

Or, how should I set "confirmed": true in "Build Downlink"?

altishchenko, 2021-11-04:
Hi! Did this solve your problem? Can you close the issue, please?

closed 2021-11-16 by gotthardp