gotthardp/lorawan-server #806: CRASH V 0.6.7 on windows server 20212

ihmp, 2021-10-29:
Hello,

I am using version 0.6.7 on windows server 2012, everything has been working fine for a very long time.
But as of today the service has not started.

I have the otp_win64_21 version

I haven't made any changes.

Thank you for your help because all the service is stopped

Thank you

Here is the file crash.log file
2021-10-30 08:06:48 =CRASH REPORT====
  crasher:
    initial call: application_master:init/4
    pid: <0.281.0>
    registered_name: []
    exception exit: {{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{timeout,[rxframe]}},[{lorawan_db,ensure_table,3,[{file,"/root/tmp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_db.erl"},{line,117}]},{lists,foreach,2,[{file,"lists.erl"},{line,1338}]},{lorawan_app,start,2,[{file,"/root/tmp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,18}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,277}]}]}}}},[{application_master,init,4,[{file,"application_master.erl"},{line,138}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
    ancestors: [<0.280.0>]
    message_queue_len: 1
    messages: [{'EXIT',<0.282.0>,normal}]
    links: [<0.280.0>,<0.42.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 987
    stack_size: 27
    reductions: 197
  neighbours:
2021-10-30 08:17:09 =CRASH REPORT====
  crasher:
    initial call: application_master:init/4
    pid: <0.286.0>
    registered_name: []
    exception exit: {{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{timeout,[rxframe]}},[{lorawan_db,ensure_table,3,[{file,"/root/tmp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_db.erl"},{line,117}]},{lists,foreach,2,[{file,"lists.erl"},{line,1338}]},{lorawan_app,start,2,[{file,"/root/tmp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,18}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,277}]}]}}}},[{application_master,init,4,[{file,"application_master.erl"},{line,138}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
    ancestors: [<0.285.0>]
    message_queue_len: 1
    messages: [{'EXIT',<0.287.0>,normal}]
    links: [<0.285.0>,<0.42.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 987
    stack_size: 27
    reductions: 197
  neighbours:
2021-10-30 08:17:10 =ERROR REPORT====
Error in process <0.236.0> on node lorawan@ns554698 with exit value:
{badarg,[{ets,select,[{#Ref<0.70215724.2653552643.110179>,282161,1000,#Ref<0.70215724.2653552644.112297>,[],0}],[]},{mnesia_lib,db_chunk,2,[{file,"mnesia_lib.erl"},{line,1131}]},{mnesia_log,ets2dcd,3,[{file,"mnesia_log.erl"},{line,934}]},{mnesia_log,ets2dcd,2,[{file,"mnesia_log.erl"},{line,921}]},{mnesia_loader,do_get_disc_copy2,4,[{file,"mnesia_loader.erl"},{line,78}]},{mnesia_controller,disc_load_table,3,[{file,"mnesia_controller.erl"},{line,2233}]},{mnesia_controller,'-load_and_reply/2-fun-0-',2,[{file,"mnesia_controller.erl"},{line,2139}]}]}

Debug.log
2021-10-30 08:06:44.099 [debug] <0.114.0>@lager_handler_watcher:126 Lager installed handler {lager_file_backend,"debug.log"} into lager_event
2021-10-30 08:06:44.099 [debug] <0.123.0>@lager_handler_watcher:126 Lager installed handler error_logger_lager_h into error_logger
2021-10-30 08:06:44.099 [debug] <0.117.0>@lager_handler_watcher:126 Lager installed handler {lager_file_backend,"error.log"} into lager_event
2021-10-30 08:06:44.443 [debug] <0.110.0>@lager_handler_watcher:126 Lager installed handler lager_backend_throttle into lager_event
2021-10-30 08:06:46.240 [debug] <0.282.0>@lorawan_app:start:17 Using config: [{max_lost_after_reset,10},{trim_interval,3600},{ssl_options,[]},{frames_before_adr,50},{http_extra_headers,#{}},{connectors,[{lorawan_connector_amqp,[<<"amqp">>,<<"amqps">>]},{lorawan_connector_mqtt,[<<"mqtt">>,<<"mqtts">>]},{lorawan_connector_http,[<<"http">>,<<"https">>]},{lorawan_connector_mongodb,[<<"mongodb">>]},{lorawan_connector_ws,[<<"ws">>]}]},{http_admin_redirect_ssl,true},{http_admin_listen,[{port,8080}]},{devstat_gap,{432000,96}},{http_custom_web,[]},{http_admin_listen_ssl,[{port,8443},{certfile,"cert.pem"},{cacertfile,"cacert.pem"},{keyfile,"key.pem"}]},{websocket_timeout,3600000},{http_admin_path,<<"/admin">>},{deduplication_delay,200},{slack_server,{"slack.com",443}},{retained_rxframes,50},{applications,[{<<"semtech-mote">>,lorawan_application_semtech_mote}]},{event_lifetime,86400},{packet_forwarder_listen,[{port,1680}]},{server_stats_interval,60},{http_admin_credentials,{<<"admin">>,<<"admin">>}},{gateway_delay,200}]
2021-10-30 08:06:48.825 [error] <0.281.0>@lorawan_db:ensure_table:117 CRASH REPORT Process <0.281.0> with 0 neighbours exited with reason: no match of right hand value {timeout,[rxframe]} in lorawan_db:ensure_table/3 line 117 in application_master:init/4 line 138
2021-10-30 08:06:48.825 [info] <0.42.0>@lorawan_db:ensure_table:117 Application lorawan_server exited with reason: no match of right hand value {timeout,[rxframe]} in lorawan_db:ensure_table/3 line 117
2021-10-30 08:06:48.825 [info] <0.42.0> Application erlmongo exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application emqttc exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application amqp_client exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application rabbit_common exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application recon exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application xmerl exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application gen_smtp exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application iso8601 exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application jsx exited with reason: stopped
2021-10-30 08:06:48.825 [info] <0.42.0> Application gun exited with reason: stopped
[... 21 lines cut ...]
2021-10-30 08:17:09.929 [info] <0.42.0> Application recon exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application xmerl exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application gen_smtp exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application iso8601 exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application jsx exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application gun exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application cowboy exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application ranch exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application ssl exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application public_key exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application asn1 exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application cowlib exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application crypto exited with reason: stopped
2021-10-30 08:17:09.929 [info] <0.42.0> Application eid exited with reason: stopped
2021-10-30 08:17:10.195 [error] emulator Error in process <0.236.0> on node lorawan@ns554698 with exit value:
{badarg,[{ets,select,[{#Ref<0.70215724.2653552643.110179>,282161,1000,#Ref<0.70215724.2653552644.112297>,[],0}],[]},{mnesia_lib,db_chunk,2,[{file,"mnesia_lib.erl"},{line,1131}]},{mnesia_log,ets2dcd,3,[{file,"mnesia_log.erl"},{line,934}]},{mnesia_log,ets2dcd,2,[{file,"mnesia_log.erl"},{line,921}]},{mnesia_loader,do_get_disc_copy2,4,[{file,"mnesia_loader.erl"},{line,78}]},{mnesia_controller,disc_load_table,3,[{file,"mnesia_controller.erl"},{line,2233}]},{mnesia_controller,'-load_and_reply/2-fun-0-',2,[{file,"mnesia_controller.erl"},{line,2139}]}]}
2021-10-30 08:17:10.304 [info] <0.42.0> Application mnesia exited with reason: stopped

Error.log

2021-10-30 08:06:48.825 [error] <0.281.0>@lorawan_db:ensure_table:117 CRASH REPORT Process <0.281.0> with 0 neighbours exited with reason: no match of right hand value {timeout,[rxframe]} in lorawan_db:ensure_table/3 line 117 in application_master:init/4 line 138
2021-10-30 08:17:05.851 [error] <0.116.0>@lager_handler_watcher:130 Lager fatally failed to install handler lager_console_backend into lager_event, NOT retrying: old_shell
[... cut ...]
{badarg,[{ets,select,[{#Ref<0.70215724.2653552643.110179>,282161,1000,#Ref<0.70215724.2653552644.112297>,[],0}],[]},{mnesia_lib,db_chunk,2,[{file,"mnesia_lib.erl"},{line,1131}]},{mnesia_log,ets2dcd,3,[{file,"mnesia_log.erl"},{line,934}]},{mnesia_log,ets2dcd,2,[{file,"mnesia_log.erl"},{line,921}]},{mnesia_loader,do_get_disc_copy2,4,[{file,"mnesia_loader.erl"},{line,78}]},{mnesia_controller,disc_load_table,3,[{file,"mnesia_controller.erl"},{line,2233}]},{mnesia_controller,'-load_and_reply/2-fun-0-',2,[{file,"mnesia_controller.erl"},{line,2139}]}]}

altishchenko, 2021-11-01:
Hello! Looks like your database got corrupted.
What you can try, is:
1. Go to the directory where your database is located.
2. Start erlang with `erl -sname lorawan`
3. Issue mnesia:start() and watch out for any error messages...

If nothing happens, you can try issuing `mnesia:wait_for_tables([<Table>], 2000).` Where `<Table>` is: config, user, server, rxframe, event ... . Possibly you will stumble over the broken one. If you will need a full list of tables, let me know.

ihmp, 2021-11-01:
Hello Alexander,

thanks a lot for your answer.

But I am really not an expert in this field, as I wrote I am under windows.

The database directory is lorawan-server/Mnesia.lorawan@ns554698

I do not understand well, can you give me a complete procedure is precise.

thank you in advance for your help.

Francois.

altishchenko, 2021-11-01:
Ok,
1. Stop (or don't start) the server.
2. Start windows command prompt or power shell
3. `cd lorawan-server`
4. `erl -sname lorawan`

You should see something like this:

```
Erlang/OTP 23 [erts-11.2.2.2] [source] [64-bit] [smp:2:2] [ds:2:2:10] [async-threads:1]

Eshell V11.2.2.2  (abort with ^G)
(lorawan@cwp-loradev)1>
```

5. `(lorawan@cwp-loradev)1> mnesia:start().` And see if there will be any error messages.
6. If there were no errors in the previous step, do the following and see which table will give you an error:

```
(lorawan@cwp-loradev)2> Tabs = mnesia:system_info(tables).
[network,config,pending,queued,area,schema,gateway,node,
 ignored_node,handler,device,profile,event,multicast_channel,
 rxframe,user,server,connector,group]
(lorawan@cwp-loradev)3> [ mnesia:wait_for_tables([T], 5000) || T <- Tabs ].
```

ihmp, 2021-11-01:
Hello,

The command erl -sname lorawan is not recognized:

C:\Windows\system32>cd..

C:\Windows>cd..

C:\>cd lorawan-server

C:\lorawan-server>erl -sname lorawan
'erl' n'est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

ihmp, 2021-11-01:
Ok I made the order in:
C: \ Program Files \ erl10.0.1 \ bin> erl -sname lorawan
Eshell V10.0.1 (abort with ^ G)
(lorawan @ ns554698) 1>

I will continue

ihmp, 2021-11-01:
I just run the commands:
C: \ Program Files \ erl10.0.1 \ bin> erl -sname lorawan
Eshell V10.0.1 (abort with ^ G)
(lorawan @ ns554698) 1> mnesia: start ().
OK
(lorawan @ ns554698) 2> mnesia: system_info (rxframe)

for (lorawan @ ns554698) 2> mnesia: system_info (rxframe) I have no return

ihmp, 2021-11-01:
Hello,
what exact order should I send?
because I have no feedback after these commands:

(lorawan @ ns554698) 2> mnesia: system_info (rxframe)

Tabs = mnesia:system_info(tables).[network,config,pending,queued,area,schema,gateway,node,ignored_node,handler,device,profile,event,multicast_channel,rxframe,user,server,connector,group]

but I am surely making a mistake on the order

thank you for your help.

altishchenko, 2021-11-01:
I see. In Erlang every statement should be ended with a dot, otherwise you won't get any feedback. Have a look at the two commands I gave above and notice the dots at their end (I put spaces to show):

```
(lorawan@cwp-loradev)2> Tabs = mnesia:system_info(tables) .
(lorawan@cwp-loradev)3> [ mnesia:wait_for_tables([T], 5000) || T <- Tabs ] .
```

ihmp, 2021-11-01:
ok je vais essayer.

pour info je suis dans une region ou il y a + 9 heures par rapport à la France (+ 11UTC)

ihmp, 2021-11-01:
here is the feedback I got for:

(lorawan @ ns554698) 3> Tabs = mnesia: system_info (tables).
[diagram]
(lorawan @ ns554698) 4>

ihmp, 2021-11-01:
for:
(lorawan@ns554698)7> [ mnesia:wait_for_tables([T], 5000) || T <- Tabs ] .
[ok]
(lorawan@ns554698)8>

ihmp, 2021-11-01:
Hello,

for controlling:

Tabs = mnesia: system_info (tables).

need to specify the tables? for example rxframe, if so how do we write the command

thank you for your help.

I can give you access to the server if you want it may be easier, you can contact me by email if needed for info.

Thank you

ihmp, 2021-11-01:
Hello,
Here is the result of the commands:

C:\Program Files\erl10.0.1\bin>erl -sname lorawan
Eshell V10.0.1  (abort with ^G)
(lorawan@ns554698)1> mnesia:start().
ok
(lorawan@ns554698)2> Tabs = mnesia:system_info(tables) .
[schema]
(lorawan@ns554698)3> [ mnesia:wait_for_tables([T], 5000) || T <- Tabs ] .
[ok]
(lorawan@ns554698)4>

but the server does not restart.
I'll admit that I'm stuck and don't know what to do.

Thank you for your help.

altishchenko, 2021-11-02:
The database directory name looks ok - `Mnesia.lorawan@ns554698`, and if you are running
`erl -sname lorawan` from your `lorawan-server/` directory it should pick up the tables from it.

From your last output I see that you are running `erl -sname lorawan` from its default directory, not the directory of the server. Please, run this command from the directory of your server.

ihmp, 2021-11-02:
Hello, when I run the command in the directory:

C: \ lorawan-server> erl -sname lorawan
'erl' is not recognized as an internal command
or external, an executable program or a batch file.

it does not recognize the erl command
I also tried in C: \ lorawan-server \ bin

my lorawan V 0.6.7 server is installed in C: \ lorawan-server

 Répertoire de C:\lorawan-server

30/10/2021  08:06    <DIR>          .
30/10/2021  08:06    <DIR>          ..
30/10/2021  08:05    <DIR>          bin
02/11/2021  05:09            71 030 crash.log
02/11/2021  05:09            24 172 debug.log
02/11/2021  05:09           932 949 erl_crash.dump
02/11/2021  05:09            10 526 error.log
30/10/2021  08:05    <DIR>          lib
02/11/2021  05:09    <DIR>          Mnesia.lorawan@ns554698
30/10/2021  08:05    <DIR>          releases
               4 fichier(s)        1 038 677 octets
               6 Rép(s)  1 750 661 922 816 octets libres

altishchenko, 2021-11-02:
Look, have you ever heard about `Environment variables`?
Google up - 'How to change PATH on Windows'. Once there, add directory `C:\Program Files\erl10.0.1\bin` to the `PATH` variable, logout and login again. After this procedure you will be able to run `erl` command from anywhere in your system, especially from the `lorawan-server\` directory.

ihmp, 2021-11-02:
Hello,
here it is 4h00
ok i just changed the path is to do the commands:

C:\lorawan-server>erl -sname lorawan
Eshell V10.0.1  (abort with ^G)
(lorawan@ns554698)1> mnesia:start().
ok
(lorawan@ns554698)2> Tabs = mnesia:system_info(tables) .
[network,config,pending,queued,txframes,handlers,area,
 schema,gateway,node,ignored_node,handler,device,profile,
 event,multicast_channel,rxframe,user,server,connector,group]
(lorawan@ns554698)3> [ mnesia:wait_for_tables([T], 5000) || T <- Tabs ] .
[ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,ok,
 ok,ok]
(lorawan@ns554698)4>

looks like everything is OK, I restart the Lorawan server by launching server-lorawan.bat?

Thank you for you help

altishchenko, 2021-11-02:
Yes, delete old log files first and erl_crash.dump. If the server fails to start, send me the erl_crash.dump file please.

ihmp, 2021-11-02:
it starts up but it is very slow, and the web page does not appear, I will see if the messages are sent correctly.
But the web display is very slow.

ihmp, 2021-11-02:
the web page: admin # / gateways / list is displayed

but the page:
/ admin # / dashboard. not displayed message: Not Found
The page you are looking for cannot be found. Take a break before trying again.

admin # / rxframes / list is not displayed
admin # / networks / list it is OK
admin # / profiles / list it is OK
admin # / devices / list it is OK
admin # / nodes / list it is OK
admin # / handlers / list it is OK
admin # / connectors / list it is OK

is there a reason for this slowness?

Thanks again for your help

altishchenko, 2021-11-02:
There should be a reason there. Possibly error.log or debug.log may help here.

altishchenko, 2021-11-02:
Basically, it looks like there is a certain problem with the `rxframe` table. When the dashboard is loaded this table is also consulted. It is not important, but you can give it a try:

1. Stop the server(!!!)
2. Start Command Prompt and go to the directory of the server (as you did previously).
3. Run `erl -sname lorawan`
4. Issue these commands:

```
(lorawan@ns554698)1> mnesia:start().
(lorawan@ns554698)2> mnesia:delete_table(rxframe).
```

5. Then close this window and start the server.

ihmp, 2021-11-02:
Error.log

2021-11-03 04:51:44.406 [error] <0.417.0> device 08817558C9ED4D6C unknown_deveui
2021-11-03 04:51:44.406 [error] <0.418.0> device 000175584167656E unknown_deveui
2021-11-03 04:52:23.345 [error] <0.480.0> handler 4350542D454C45432D4D4F4E4F {parse_failed,{error,function_clause}}
2021-11-03 04:52:57.193 [error] <0.497.0> device 0004A30B00ED8C32 unknown_deveui
2021-11-03 04:53:17.632 [error] <0.504.0> device 000175584167656E unknown_deveui
2021-11-03 04:54:29.096 [error] <0.640.0> device 06AB735E2167656E unknown_deveui
2021-11-03 04:54:29.104 [error] <0.641.0> device 000175584167656E unknown_deveui
2021-11-03 04:55:24.917 [error] <0.703.0> device 000170344167656E unknown_deveui
2021-11-03 04:56:49.567 [error] <0.791.0> device 0004A30B00ED8C32 unknown_deveui
2021-11-03 04:56:59.146 [error] <0.808.0> device 000175584167656E unknown_deveui
2021-11-03 04:57:06.490 [error] <0.815.0> device 4883C7DF30051A3A unknown_deveui
2021-11-03 04:57:08.350 [error] <0.823.0> device 0004A30B00ED8C32 unknown_deveui
2021-11-03 04:59:26.911 [error] <0.976.0> device 000170344167656E unknown_deveui
2021-11-03 04:59:54.493 [error] <0.1041.0> device 70B3D5705000220B unknown_deveui
2021-11-03 05:01:23.768 [error] <0.1115.0> device 000170344167656E unknown_deveui
2021-11-03 05:01:51.411 [error] <0.1128.0> device 0004A30B00ED8E04 unknown_deveui
2021-11-03 05:02:19.586 [error] <0.1147.0> device 0004A30B00ED8E04 unknown_deveui
2021-11-03 05:02:29.665 [error] <0.1160.0> device 9091E558D167656E unknown_deveui
2021-11-03 05:02:29.665 [error] <0.1161.0> device 000175584167656E unknown_deveui
2021-11-03 05:05:33.170 [error] <0.1303.0> device 000170344167656E unknown_deveui
2021-11-03 05:07:51.763 [error] <0.1400.0> device 000175584167656E unknown_deveui
2021-11-03 05:10:26.445 [error] <0.1639.0> device 000170344167656E unknown_deveui
2021-11-03 05:13:36.703 [error] <0.1833.0> device 000170344167656E unknown_deveui
2021-11-03 05:19:04.259 [error] <0.2079.0> device 000170564167656E unknown_deveui
2021-11-03 05:23:26.548 [error] <0.2328.0> device 000175584167656E unknown_deveui
2021-11-03 05:23:52.443 [error] <0.2352.0> device 000175584167656E unknown_deveui
2021-11-03 05:24:08.477 [error] <0.2393.0> device 70B3D57050002256 unknown_deveui
2021-11-03 05:24:45.795 [error] <0.2443.0> device C9AE70571757656E unknown_deveui
2021-11-03 05:25:20.801 [error] <0.2507.0> device 70B3D57050000CE5 unknown_deveui
2021-11-03 05:26:18.145 [error] <0.2724.0> device 000175584167656E unknown_deveui
2021-11-03 05:27:04.165 [error] <0.2777.0> device 000175584167656E unknown_deveui
2021-11-03 05:29:31.537 [error] <0.3035.0> device 000175584167656E unknown_deveui
2021-11-03 05:31:28.405 [error] <0.3135.0> device 000175584167656E unknown_deveui
2021-11-03 05:36:24.580 [error] <0.3369.0> device 000175584077657E unknown_deveui
2021-11-03 05:36:53.412 [error] <0.3395.0> device 70B3D570500020E0 unknown_deveui
2021-11-03 05:38:51.042 [error] <0.3466.0> device 000175584167656E unknown_deveui
2021-11-03 05:40:42.781 [error] <0.3573.0> device 000170344167656E unknown_deveui
2021-11-03 05:42:26.149 [error] <0.3680.0> device 000175584167656E unknown_deveui
2021-11-03 05:42:46.761 [error] <0.3692.0> device 000170344167656E unknown_deveui
2021-11-03 05:43:55.045 [error] <0.3740.0> device 000175584167656E unknown_deveui
2021-11-03 05:44:51.418 [error] <0.3778.0> device 000170344167656E unknown_deveui
2021-11-03 05:46:46.478 [error] <0.3866.0> device 000175584167656E unknown_deveui
2021-11-03 05:48:33.535 [error] <0.3958.0> device 000170344167656E unknown_deveui
2021-11-03 05:48:45.126 [error] <0.3973.0> device 70B3D57050002156 unknown_deveui
2021-11-03 05:52:17.772 [error] <0.4755.0> device 000175584167656E unknown_deveui
2021-11-03 05:53:54.161 [error] <0.4820.0> device 000170344167656E unknown_deveui
2021-11-03 05:55:53.023 [error] <0.4916.0> device 000170344167656E unknown_deveui
2021-11-03 05:56:18.105 [error] <0.4933.0> device 4883C7DF30042308 unknown_deveui
2021-11-03 05:56:24.394 [error] <0.4935.0> device 70B3D57050002260 unknown_deveui
2021-11-03 05:57:49.209 [error] <0.5014.0> device 000170344167656E unknown_deveui
2021-11-03 05:58:37.739 [error] <0.5049.0> device 0004A30B00ED8C32 unknown_deveui
2021-11-03 06:01:51.501 [error] <0.5187.0> device 0004A30B00ED8E04 unknown_deveui
2021-11-03 06:02:00.898 [error] <0.5197.0> device 0004A30B00ED8E04 unknown_deveui

ihmp, 2021-11-02:
[debug.log](https://github.com/gotthardp/lorawan-server/files/7463445/debug.log)

ihmp, 2021-11-02:
IT IS OK 😁

C: \ lorawan-server> erl -sname lorawan
Eshell V10.0.1 (abort with ^ G)
(lorawan @ ns554698) 1> mnesia: start ().
OK
(lorawan @ ns554698) 2> mnesia: delete_table (rxframe).
{atomic, ok}
(lorawan @ ns554698) 3>

Thank you very much Alexander, now I can see all the web pages and the server is OK
Thanks for this great job.

I will do a procedure if it starts again 😁
Thank you.

ihmp, 2021-11-02:
one last question, but maybe we need to do another post.

Following this problem, is there a way to save all the configuration of the Lorawan server, in order to use this backup on a new installation of the server in the event of a problem.
If yes is there a detailed procedure

Thank you.

altishchenko, 2021-11-02:
Oh yes. There is a `dbexport.bat` file in the server `bin/` directory, that will export all of the relevant server configuration into .json files, which you may later use with `dbimport.bat` utility to configure new server. You may want to edit this file, in case you changed the default login/password for the server.

ihmp, 2021-11-03:
Ok i will do it for safety

and again thank you very much for your help.

altishchenko, 2021-11-04:
Hi! Can you, please, close the issue if it is fine with you?

ihmp, 2021-11-04:
ok, and thank you again

closed 2021-11-04 by ihmp