gotthardp/lorawan-server #807: Can't connect to MQTT (Mosquitto)

horvatmiha, 2021-11-04:
I have a problem connecting to MQTT mosquitto broker. I am running mosquitto on local machine. I've configured mosquitto.conf file correctly, I am able to connect to broker successfully with mqtt explorer.

In debug.log I got these lines:
`2021-11-04 12:03:22.314 [debug] <0.2815.0>@lorawan_connector_mqtt:reconnect:142 Reconnecting
2021-11-04 12:03:22.314 [warning] <0.2876.0> ClientId is NULL!
2021-11-04 12:03:22.314 [info] <0.2876.0> [Client <0.2815.0>]: connecting to 192.168.1.216:1883
2021-11-04 12:03:22.316 [info] <0.2876.0> [emqttc_MIHA-PC_70578186f6ac06b6f775@192.168.1.216:1904] SENT: CONNECT(Q0, R0, D0, ClientId=emqttc_MIHA-PC_70578186f6ac06b6f775, ProtoName=MQTT, ProtoVsn=4, CleanSess=true, KeepAlive=0, Username=undefined, Password=undefined)
2021-11-04 12:03:22.316 [info] <0.2876.0> [emqttc_MIHA-PC_70578186f6ac06b6f775@192.168.1.216:1904] SENT: <<16,47,0,4,77,81,84,84,4,2,0,0,0,35,101,109,113,116,116,99,95,77,73,72,65,45,80,67,95,55,48,53,55,56,49,56,54,102,54,97,99,48,54,98,54,102,55,55,53>>
2021-11-04 12:03:22.316 [info] <0.2876.0> [Client <0.2815.0>] connected with 192.168.1.216:1883
2021-11-04 12:03:22.316 [info] <0.2876.0> [Client <0.2815.0>] RECV: CONNACK_INVALID_ID
2021-11-04 12:03:22.316 [debug] <0.2815.0>@lorawan_connector_mqtt:handle_info:242 Connector <<"Testni_connector">> to <<"mqtt://192.168.1.216:1883">> (attempt311) failed: {shutdown,{connack_error,'CONNACK_INVALID_ID'}} (count: 12)`

[image]

[image]

Any help would be great!
Thanks in advance!

altishchenko, 2021-11-04:
Warning! By **standard** only these characters are allowed in the Client Id:

0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ

Mosquitto may allow other characters, but not the ones that go into auto-generated Id.

horvatmiha, 2021-11-05:
I filled Client Id with "lorawanServer" which according to allowed characters is OK. Here is my auth tab:
[image]

I am still getting exactly the same error in debug.log. Please take a look at the attached file.
[debug.log](https://github.com/gotthardp/lorawan-server/files/7485966/debug.log)

I've tried publishing message from Node-Red. Works like a charm. Take a look at the configuration:
[image]
[image]
[image]
[image]

MQTT explorer receiving message from Node-Red over MQTT:
[image]

altishchenko, 2021-11-05:
And do you have anything in mosquitto logs about these connection attempts?

horvatmiha, 2021-11-05:
This is what I got in mosquitto logs:
`2021-11-05T18:45:46: New connection from 192.168.1.216:2353 on port 1883.
2021-11-05T18:45:46: New client connected from 192.168.1.216:2353 as lorawanServer (p1, c1, k0).
2021-11-05T18:45:46: Bad socket read/write on client lorawanServer: Invalid arguments provided.`

Also take a look at the full mosquitto.log file.
[mosquitto.log](https://github.com/gotthardp/lorawan-server/files/7487968/mosquitto.log)

horvatmiha, 2021-11-05:
As referenced in issue #796 downgrading from 2.0.12 to 2.0.11 solved the problem. Connector is now successfully connected to the broker with the same parameters in all tabs as above. Also with the same mosquitto.conf file.

[image]

Thanks for help!

altishchenko, 2021-11-05:
Oh, I missed that one. Sorry.

horvatmiha, 2021-11-05:
@altishchenko I am closing this issue. Thank you for your help!

closed 2021-11-05 by horvatmiha