gotthardp/lorawan-server #808: Multiple Downlinks - Wrong values

jot4p, 2021-11-10:
Hi,

I need some help with downlinks. 

I'm receiving, for all sensors, the value of the uplink in the downlink and, in addition, the values are repeated 4 times.

Packets are arriving via MQTT via from multitech's packet forwareder.

Any idea why this happen ?

[image]

altishchenko, 2021-11-11:
Hi, it is not clear from your description, how the downlinks are originally generated. Can you share your Handler and Connector config?

jot4p, 2021-11-11:
Hello @altishchenko .

Problem solved.. 

PROBLEM: I'm subscriving the same topic for downlink and uplink...    thanks for the help.

[image]