gotthardp/lorawan-server #817: Gateway showing disconnected

Sonia-Hmidi, 2022-03-29:
Hello,
i have installed the server on my virtual machine (ubuntu 20.04 release). i am using the LoRa\LoRaWAN Gateway-868MHz with RPi3.
i've changed the server ip @ to the VM ip @ in the local_conf.json but my gateway is showing status disconnected.

Sonia-Hmidi, 2022-04-01:
Update : when i installed the server on my windows machine the previous problem is solved but another issue appeared. When i send an uplink message an error message **ack_lost** appears. As shown in the figure bellow, in the frames it shows me the values of SNR, RSSI... but not the actual data. I tried to increase the push_timout_ms value but the issue wasn't solved.Does this mean that the message was lost? and if so how to solve this issue? @gotthardp 
[image]

sn3kyJ3di, 2022-04-17:
Is this happening on one sensor or all?

fthiebolt, 2022-04-29:
We're facing a similar issue with gateways not belonging to our internal network. Right now, we're using a docker container whose 172.17.0.1 is the docker0 bridge. Even gateways featuring a public IP are told to have 172.17.0.1 as address: when you have only one, it works but if you add a second one (like here), multiple disconnect appeared :| 
[image]