gotthardp/lorawan-server #837: Startup Error

agilecomputing, 2023-04-28:
Hello,

I am having a startup error when trying to run the lorawan-server. I have tried to install it both as a service and just as an executable.

The error seems to be like one of the previous posts where the .beam file is not being found, but I cannot figure out why.

When I run lorawan-server, the error I get is:

{"init terminating in do_boot",{undef,[{lorawan_app,start,[],[]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
init terminating in do_boot ({undef,[{lorawan_app,start,[],[]},{init,start_em,1,[]},{init,do_boot,3,[]}]})

Crash dump is being written to: erl_crash.dump...done

I am using Erlang/OTP version 25. Output when I run erl is:
Erlang/OTP 25 [erts-13.1.4] [source] [64-bit] [smp:1:1] [ds:1:1:10] [async-threads:1] [jit:ns]

Eshell V13.1.4  (abort with ^G)
1>

All of this is running on an Ubuntu Server x64 18.0.4.5. Output of uname -a is:
Linux lora-server 4.15.0-210-generic #221-Ubuntu SMP Tue Apr 18 08:32:52 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

I must be missing something simple - any help / pointers you can provide would be greatly appreciated.

Thank you.

BastienV-SATIE, 2023-04-29:
Hello,

I have the same problem here :
$ bin/lorawan-server
[os_mon] memory supervisor port (memsup): Erlang has closed
[os_mon] cpu supervisor port (cpu_sup): Erlang has closed
{"init terminating in do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{error,eaddrinuse}},[{lorawan_app,start_http,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,88}]},{lorawan_app,start,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,23}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,293}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,13}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
init terminating in do_boot ({{badmatch,{error,{lorawan_server,{_}}}},[{lorawan_app,start,0,[{_},{_}]},{init,start_em,1,[]},{init,do_boot,3,[]}]})

Crash dump is being written to: erl_crash.dump...done

Erlang version : 
lorawan-server$ erl
Erlang/OTP 24 [erts-12.2.1] [source] [64-bit] [smp:2:2] [ds:2:2:10] [async-threads:1] [jit]

Eshell V12.2.1  (abort with ^G)

Best regards!

BastienV-SATIE, 2023-04-29:
Everything working here : https://github.com/abelino/lorawan-server/tree/otp25-support

agilecomputing, 2023-05-01:
Did you download and build this version yourself?

Thank you.

BastienV-SATIE, 2023-05-01:
Yes i have with erlang/otp 25.